coolsnowwolf/lede #1594: 麻烦老大替换一下ssrurl，这样可以导入三种链接。

wuypinfo, 2019-09-02:
我是在其他人的固件里分离得到的。

[导入三种链接.zip](https://github.com/coolsnowwolf/lede/files/3564174/default.zip)

shikkoku, 2019-09-02:
完全不明白你再说什么

wuypinfo, 2019-09-02:
[image]
大多数固件编译出来只支持ssr的链接倒入，少数可以倒入三种。虽然可以订阅，你有没有发现，删除订阅死链接很恼人？

shikkoku, 2019-09-03:
[image]
你是指希望能在SSR PLUS+的新建节点这里增加导入ss、vmess类型链接导入是吗，麻烦描述清楚点。

wuypinfo, 2019-09-03:
我希望能在SSR PLUS+的新建节点这里增加导入ss、vmess类型链接导入。
[导入三种链接.zip](https://github.com/coolsnowwolf/lede/files/3567525/default.zip)
这里有2个ssrurl，mix的是可以匹配目前该源文件。另外一个是潘多拉固件分离的，你直接运行就能看到。如果放入路由，去掉mix，还原为ssrurl.htm

adamw92, 2019-09-03:
> 我希望能在SSR PLUS+的新建节点这里增加导入ss、vmess类型链接导入。
> [导入三种链接.zip](https://github.com/coolsnowwolf/lede/files/3567525/default.zip)
> 这里有2个ssrurl，mix的是可以匹配目前该源文件。另外一个是潘多拉固件分离的，你直接运行就能看到。如果放入路由，去掉mix，还原为ssrurl.htm

感谢~直接替换mix的版本之后SS可以导入了，目前导入正常，方便了不少…

closed 2019-09-05 by wuypinfo