coolsnowwolf/lede #1597: V2ray的订阅加密方式字段不正确

MoeGrid, 2019-09-02:
`/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.sh`文件第163行，
使用的是json中的type作为v2ray的加密方式，实际上根据 [分享链接格式说明(ver 2)](https://github.com/2dust/v2rayN/wiki/%E5%88%86%E4%BA%AB%E9%93%BE%E6%8E%A5%E6%A0%BC%E5%BC%8F%E8%AF%B4%E6%98%8E(ver-2)) 这个文档，type字段是用来放伪装类型的，并不是加密方式，这导致订阅出来的节点加密方式全都是默认的none，连接不上。建议将默认改为auto

zxlhhyccc, 2019-09-03:
> `/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.sh`文件第163行，
> 使用的是json中的type作为v2ray的加密方式，实际上根据 [分享链接格式说明(ver 2)](https://github.com/2dust/v2rayN/wiki/%E5%88%86%E4%BA%AB%E9%93%BE%E6%8E%A5%E6%A0%BC%E5%BC%8F%E8%AF%B4%E6%98%8E(ver-2)) 这个文档，type字段是用来放伪装类型的，并不是加密方式，这导致订阅出来的节点加密方式全都是默认的none，这样是连接不上的。建议将加密方式直接定死为auto，v2ray会自动选择合适的加密方式。

你修改了尝试过吗？

MoeGrid, 2019-09-03:
尝试过了，没问题

MoeGrid, 2019-09-06:
> > `/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.sh`文件第163行，
> > 使用的是json中的类型作为v2ray的加密方式，实际上根据[分享链接格式说明（ver 2）](https://github.com/2dust/v2rayN/wiki/%E5%88%86%E4%BA%AB%E9%93%BE%E6%8E%A5%E6%A0%BC%E5%BC%8F%E8%AF%B4%E6%98%8E(ver-2))这个文档，type字段是用来放伪装类型的，并不是加密方式，这导致订阅出来的节点加密方式全都是默认的没有，这样是连接不上的。建议将加密方式直接定死为汽车，v2ray会自动选择合适的加密方式。
> 
> 你修改了尝试过吗？

尝试过了，没问题

closed 2019-09-29 by coolsnowwolf