coolsnowwolf/lede #1601: ssr plus+的守护进程问题

HiGarfield, 2019-09-03:
尝试过，只要ssr plus+中用到kcptun，当kcptun挂了（进程还在，无法正常通讯）的情况下，守护进程不会自动重启kcptun。

zxlhhyccc, 2019-09-03:
> 尝试过，只要ssr plus+中用到kcptun，当kcptun挂了（进程还在，无法正常通讯）的情况下，守护进程不会自动重启kcptun。

临时解决办法就是按照luci-app-kcptun解决。。。

HiGarfield, 2019-09-03:
> > 尝试过，只要ssr plus+中用到kcptun，当kcptun挂了（进程还在，无法正常通讯）的情况下，守护进程不会自动重启kcptun。
> 
> 临时解决办法就是按照luci-app-kcptun解决。。。

具体是怎么做的？谢谢！

zxlhhyccc, 2019-09-04:
> > > 尝试过，只要ssr plus+中用到kcptun，当kcptun挂了（进程还在，无法正常通讯）的情况下，守护进程不会自动重启kcptun。
> > 
> > 
> > 临时解决办法就是按照luci-app-kcptun解决。。。
> 
> 具体是怎么做的？谢谢！

编译luci-app-kcptun进去啊

HiGarfield, 2019-09-04:
> > > > 尝试过，只要ssr plus+中用到kcptun，当kcptun挂了（进程还在，无法正常通讯）的情况下，守护进程不会自动重启kcptun。
> > > 
> > > 
> > > 临时解决办法就是按照luci-app-kcptun解决。。。
> > 
> > 
> > 具体是怎么做的？谢谢！
> 
> 编译luci-app-kcptun进去啊

意思是直接放弃SSR Plus里面的kcptun功能？

HiGarfield, 2019-09-06:
最新测试结果，用老的源码，ssr pro + luci-app-kcptun，正常工作。看来是ssr plus+中kcptun守护进程的问题了。

closed 2019-09-29 by coolsnowwolf