coolsnowwolf/lede #1607: 公司环境下SSR plus+性能问题

alechsu, 2019-09-05:
固件家用完全没有问题，但是公司用特别是ip route模式下就无法上网了，就像是堵住了一样，看了下，公司设备在250台左右，平时每天在线5000-6000条并发连接，因为是外企很大一部分链接应该是连到境外诸如office365云服务。请高手指教有没有可以调整SSR plus+插件性能的地方，或者插件的设计性能就不能满足这样的并发数？谢谢！

poemsiu, 2019-09-05:
你的路由器性能够不够，cpu占用如何

alechsu, 2019-09-05:
路由器性能不可能不够的，Dell R630 虚拟服务器

rampageX, 2019-09-05:
haproxy 做负载均衡

alechsu, 2019-09-05:
所以这是服务端的性能问题？不是SSR plus+的问题？

coolsnowwolf, 2019-09-05:
感觉是服务器端性能不足的问题，htop 截个图看看

coolsnowwolf, 2019-09-05:
如果是客户端问题，其实可以启用更多的 SS/SSR 线程数解决，例如 16/32，自动内核级负载均衡的

alechsu, 2019-09-06:
[image]

alechsu, 2019-09-06:
上面贴图是gfwlist模式下的htop，可运行，但是切换到ip route外网就断了。。。。我觉得不是机器性能的为题，可能是ip route 模式的性能问题？或者ip route有bug，大并发量下面会当？

lifetyper, 2019-09-11:
SSR服务器端都是python版的，只能利用单核CPU的性能，并发数一多，完全可能在你客户端崩掉之前服务端就挂了。

alechsu, 2019-09-12:
不太懂python，我用v2ray也是一样，所以这些翻墙方案都不是设计给公司用的？

lifetyper, 2019-09-13:
> 不太懂python，我用v2ray也是一样，所以这些翻墙方案都不是设计给公司用的？

公司不是应该拉专线？还用SSR给全公司翻墙，网管是想喝茶了？

coolsnowwolf, 2019-09-14:
服务器端的性能问题了，需要在服务器端用 tcpreuseport 解决

alechsu, 2019-09-16:
> > 不太懂python，我用v2ray也是一样，所以这些翻墙方案都不是设计给公司用的？
> 
> 公司不是应该拉专线？还用SSR给全公司翻墙，网管是想喝茶了？

因为到海外分公司的专线带宽有限，并且我也试过了从专线连接分公司的ssr服务器，gfwlist模式没啥问题，但是切换到ip route模式没几秒种就死。。。

lifetyper, 2019-09-16:
> > > 不太懂python，我用v2ray也是一样，所以这些翻墙方案都不是设计给公司用的？
> > 
> > 
> > 公司不是应该拉专线？还用SSR给全公司翻墙，网管是想喝茶了？
> 
> 因为到海外分公司的专线带宽有限，并且我也试过了从专线连接分公司的ssr服务器，gfwlist模式没啥问题，但是切换到ip route模式没几秒种就死。。。
找rixcloud定制企业版的服务吧

alechsu, 2019-09-19:
> 服务器端的性能问题了，需要在服务器端用 tcpreuseport 解决

问题最终解决了，确实是插件的并发能力被系统默认限制了，我在ssr插件的启动脚本（/etc/init.d/shadowsocksr）里面加了两条命令：ulimit -n 1000000 和 ulimit -p 65000 （怀疑只需要ulimit -n就行了）然后服务端也打开linux默认限制后，绕过大陆模式非常稳定了，另外我在设置v2ray时候用了mux参数来控制总体连到vps的连接数。
[image]

closed 2019-09-22 by alechsu