coolsnowwolf/lede #1621: 编译出现错误,与libgpg-error-1.12 有关

lan542662, 2019-09-13:
重新编译2次,make clean,dl删除重新编译还是错误,男神~~
就修改了setting zzz.添加了pppoe拨号和密码
[https://pastebin.com/S7hP7ZAy](https://pastebin.com/S7hP7ZAy)

[image]

adaaaaaa, 2019-09-17:
我也是！一毛一样的报错！！😢😢😢😢😢😢

hcym, 2019-09-17:
加芝麻开门就行了，其它熟练了再说

adaaaaaa, 2019-09-17:
@hcym  但是，我没有搞芝麻开门和其他的东西，也是这样报错哦。。。😢😢😢😢😢😢

Margatroide, 2019-09-17:
一样的问题呀，折腾一下午了

hcym, 2019-09-17:
那就是其它问题，比如网络限制，系统依赖，等等等等都是细节不起眼的细节。还有配置尽量减少，大雕的默认太多太多，我的只有翻墙功能，其它一概删除，很少出错，而且现在n1和蜗牛都是用19.07的官方版编译，那个是一天变动几次，也没什么问题

lan542662, 2019-09-18:
> 加芝麻开门就行了，其它熟练了再说

后来全部清空，git clone 原汁原味编译 继续这个错误，看来和修改没关系。而且之前系统编译通过的。

lan542662, 2019-09-18:
> 我也是！一毛一样的报错！

你是什么系统，我是archlinux，之前一点问题没有。

Margatroide, 2019-09-18:
我也是archlinux，昨天用台式新安装的试了一天不成，现在在拿笔记本的老环境测试

Margatroide, 2019-09-18:
> 
> 
> > 我也是！一毛一样的报错！
> 
> 你是什么系统，我是archlinux，之前一点问题没有。

刚用笔记本两个多月没滚过的arch编译了一遍一次过，看来很有可能是发行版问题，先用虚拟机换个发行版试试吧

lan542662, 2019-09-18:
> > > 我也是！一毛一样的报错！
> > 
> > 
> > 你是什么系统，我是archlinux，之前一点问题没有。
> 
> 刚用笔记本两个多月没滚过的arch编译了一遍一次过，看来很有可能是发行版问题，先用虚拟机换个发行版试试吧

我之前也是一遍过，看来果然是这个问题，不知道什么问题。因为没有出现依赖问题

peeweep, 2019-10-08:
只查到这一篇相关讨论 https://dev.gnupg.org/T4469 https://dev.gnupg.org/T4459
好像是gawk 5.0以后libgpg-error 才有的这个 bug, 我降级到 4.2.1-2，成功编译了

hcym, 2019-10-08:
刚刚编译了n1，没有什么情况发生，大雕的默认app只保留ss＋和ad＋，很顺利

kktt007, 2019-10-11:
> 我也是同样的报错，https://gist.github.com/peeweep/5a54574c4513e5413ea4f8fb91d3a413
> 查到这一篇相关讨论 https://dev.gnupg.org/T4469 https://dev.gnupg.org/T4459
> 好像是gawk 5.0以后libgpg-error 才有的这个 bug, 我降级到 4.2.1-2，成功编译了
谢谢提醒，我也查了下，op那边好像修复了，大屌这边没修改吧，你看看先
https://github.com/openwrt/packages/tree/master/libs/libgpg-error

hcym, 2019-10-11:
1804一直正常，纳闷儿

peeweep, 2019-10-11:
op 那边前几天还有报错的 https://forum.openwrt.org/t/45910

kktt007, 2019-10-11:
> op 那边前几天还有报错的 https://forum.openwrt.org/t/45910

这人和我一样也是R7800,难道和编译的是啥也有关系啊...(猜测

peeweep, 2019-10-11:
没什么关系吧, 我是x86。
现在已经知道的是 [Arch](https://git.archlinux.org/svntogit/packages.git/tree/trunk?h=packages/libgpg-error) [Gentoo](https://gitweb.gentoo.org/repo/gentoo.git/tree/dev-libs/libgpg-error/files) [macOS](https://github.com/Homebrew/linuxbrew-core/blob/master/Formula/gawk.rb) 出问题了, 这三个发行版都是gawk5.0 以上, 都同步了[libgpg-error上游的patch](https://dev.gnupg.org/rE7865041c77f4f7005282f10f9b6666b19072fbdf)

Margatroide, 2019-10-11:
> 
> 
> 我也是同样的报错，https://gist.github.com/peeweep/5a54574c4513e5413ea4f8fb91d3a413
> 查到这一篇相关讨论 https://dev.gnupg.org/T4469 https://dev.gnupg.org/T4459
> 好像是gawk 5.0以后libgpg-error 才有的这个 bug, 我降级到 4.2.1-2，成功编译了

我当时也搜到了这个问题，翻原版openwrt的feeds里面找到libgpg-error的编译目录看到有关于gawk5.0的patch没想太多就下载下来放进目录试着编译没过，以为和这个无关来着

kktt007, 2019-10-11:
> > 我也是同样的报错，https://gist.github.com/peeweep/5a54574c4513e5413ea4f8fb91d3a413
> > 查到这一篇相关讨论 https://dev.gnupg.org/T4469 https://dev.gnupg.org/T4459
> > 好像是gawk 5.0以后libgpg-error 才有的这个 bug, 我降级到 4.2.1-2，成功编译了
> 
> 我当时也搜到了这个问题，翻原版openwrt的feeds里面找到libgpg-error的编译目录看到有关于gawk5.0的patch没想太多就下载下来放进目录试着编译没过，以为和这个无关来着

刚换成gawk4.12了.把dl和tmp删除,然后make clean重编译,还是错误..
[Please fix ./patches/001-gcc5.patch!
](https://pastebin.com/RwQjZ1Lw)  还是错https://pastebin.com/RwQjZ1Lw

peeweep, 2019-10-11:
001-gcc5.patch 改过? 进 feeds/packages 里 git status 看一下

peeweep, 2019-10-11:
这是我在用的patch, 从op仓库拿的, 现在本地gawk是5.0.1-1, 编译通过
放在 feeds/packages 里就可以 [libgpg-error-patch.zip](https://github.com/coolsnowwolf/lede/files/3717722/libgpg-error-patch.zip)
``` shell
cd feeds/packages
git am < ~/Downloads/libgpg-error-patch/*.patch
```

kktt007, 2019-10-13:
> 这是我在用的patch, 从op仓库拿的, 现在本地gawk是5.0.1-1, 编译通过
> 放在 feeds/packages 里就可以 [libgpg-error-patch.zip](https://github.com/coolsnowwolf/lede/files/3717722/libgpg-error-patch.zip)
> 
> ```shell
> cd feeds/packages
> git am < ~/Downloads/libgpg-error-patch/*.patch
> ```

谢谢,可能确实之前改的导致有问题,我刚把gawk换成4.12 清空重新编译,好了,然后升级到gawk5.0.1-1依然可以编译了,要是再有问题,再来用你的patch文件,关闭issue咯.

closed 2019-10-13 by kktt007

kktt007, 2019-11-02:
今天编译x86固件,又来错误了
https://pastebin.com/3i3haRPj

hcym, 2019-11-02:
也是内核错误4.19＊＊＊，大头针灸，还是用老固件吧

kktt007, 2019-11-26:
> 
> 
> 这是我在用的patch, 从op仓库拿的, 现在本地gawk是5.0.1-1, 编译通过
> 放在 feeds/packages 里就可以 [libgpg-error-patch.zip](https://github.com/coolsnowwolf/lede/files/3717722/libgpg-error-patch.zip)
> 
> ```shell
> cd feeds/packages
> git am < ~/Downloads/libgpg-error-patch/*.patch
> ```

谢谢，我在Openwrt看到又3个patch，放进去，正常了。但是出现了新问题，802mac11.ko Migging 问题，然后我又用Openwrt的替换，然后还是错误，qt错误也来了。我已经放弃了。对了我用的是archlinux.
楼下改内核什么的我也试过了，没用。
然后我装了个ubuntu 18.04 然后编译，没任何问题。所以lean的固件还是用ubuntu编译吧。(如果又问题可以make clean )