coolsnowwolf/lede #1624: SSR Plus+的绕过中国大陆IP模式存在重大问题，无法过滤BT种子下载流量。

jy3344511, 2019-09-14:
首先，经过测试，使用电脑版的SSR，代理使用--全局模式，规则使用绕过局域网和大陆的情况下（软件目录下面有chn_ip的记事本文件），在使用BT软件(比特彗星、Xdown等)下载种子文件时，所有的流量--是--不--经过代理的。

问题反馈：
软路由作为旁路由，使用SSR Plus+的情况下，只有选择GFW列表模式的情况下，用BT软件下载种子文件时的流量，才不经过代理，从而保证一个月的流量够使用。如果使用绕过中国大陆IP模式，那么下载种子文件时的流量，要经过代理，这样的话，一个月的流量很快就用完了。

在使用软路由的情况下，切换节点、关闭SSR Plus+等操作，都比电脑端SSR的步骤要多一些，所以麻烦的多。而之所以不使用GFW列表模式，来防止BT的流量经过代理的原因，是因为要经常使用--搜索引擎，而搜索出来的网站，有些打不开，是网站本身已经不存在了，而更多的情况是相当一部分需要手动添加到GFW列表里，才能打开网站，这样就造成了大量的操作，导致使用SSR Plus+及其不方便，希望能改进SSR Plus+的绕过中国大陆IP模式，来防止BT的流量经过代理。

备注：使用的BT软件，并没有打开软件的代理功能。

shikkoku, 2019-09-15:
很正常，你BT下载同样会连接到海外节点和用户，所以使用绕过中国大陆模式当然会经过过SSR流量了。你电脑版的所谓全局模式只是浏览器之类能全局部分大陆海外代理，你试试封大陆IP的海外游戏能上？路由器就可以。SSR PLUS目前没有端口过滤功能，所以你本机要用SSR PLUS而且又需BT只能继续用GFW模式来规避。

yunhai20082008, 2019-09-15:
> 只是浏览器之类能全局部分大陆海外代理
这句话什么意思

yunhai20082008, 2019-09-15:
你还是用gfw模式吧，打不开的就添加，我自己添了20多个。
我感觉这设计没问题
至于你说的全局模式下绕过局域网和大陆，再用bt，流量不走国外，我感觉可能性不大，也可能你测试时下载的bt没有国外能链接的吧，具体bt啥原理我不清楚，我觉得应该是我说的样子

jy3344511, 2019-09-16:
又测试了，电脑端SSR，代理使用--全局模式，规则使用绕过局域网和大陆的情况下（软件目录下面有chn_ip的记事本文件），BT下载的流量的确不经过代理，一共下载了几十G的东西，流量不变，看几分钟的油管，再刷新，使用流量马上有变化。

另外，还发现一个很麻烦的地方，一个网站不翻墙，根本打不开。使用GFW的情况下，即便添加了新的网站进入，重启路由器。个别网站打开后，图片半天刷新不出来，选用绕过中国大陆IP模式，图片直接刷新出来，用的同一个节点。

rampageX, 2019-09-16:
网站是网站域名，图片是图片域名，1024 图床都是要单独过 GFWList 的。话说 lede 居然没使用自动过墙的 redsocks 或者 tcproute 或者 cow？

shikkoku, 2019-09-16:
> 又测试了，电脑端SSR，代理使用--全局模式，规则使用绕过局域网和大陆的情况下（软件目录下面有chn_ip的记事本文件），BT下载的流量的确不经过代理，一共下载了几十G的东西，流量不变，看几分钟的油管，再刷新，使用流量马上有变化。
> 
> 另外，还发现一个很麻烦的地方，一个网站，根本打不开。使用GFW的情况下，即便添加了新的网站进入，重启路由器。网站打开后，图片半天刷新不出来，使用绕过中国大陆IP模式，图片直接刷新出来，用的同一个节点。

你试试在BT软件里启用系统代理再用“绕过局域网和中国大陆”看看走不走？你是不明白windows客户端的工作原理吧？

coolsnowwolf, 2019-09-16:
1. 绕过中国IP就是和国外IP走代理连接，这个没有问题
2. 你知道如何过滤出来BT 的流量么（端口随机，只能靠 L7 nDPI）。分流的前提是先做协议过滤。
3. 选一台下载机，加入 LAN 白名单即可

这个功能没有任何问题

coolsnowwolf, 2019-09-16:
至于楼上说的 redsocks 或者 tcproute 或者 cow，这倒是简单了，所有 tdp/udp 流量都交给 app嘛～～这样完全不支持 qos，fast path 之类了

我不相信现在每个人都是 I9 9900K 软路由了，你说呢？

jy3344511, 2019-09-16:
那么以电脑端比特彗星为例，IP为192.168.1.120，端口为：15625，怎么在使用SSR Plus绕过中国大陆IP的模式下，让比特彗星这个软件的流量都不经过代理呢？之前就奇怪流量用的太快了，现在一直找不到方法，网上说有的代理，可以添加某一个端口的IP地址下的流量都不经过代理，这应该在哪添加，找不到啊。

404user, 2019-09-16:
> 至于楼上说的 redsocks 或者 tcproute 或者 cow，这倒是简单了，所有 tdp/udp 流量都交给 app嘛～～这样完全不支持 qos，fast path 之类了
> 
> 我不相信现在每个人都是 I9 9900K 软路由了，你说呢？

大神，能帮忙看看这个帖子吗？ https://github.com/coolsnowwolf/lede/issues/1618  拜托了，谢谢！

rampageX, 2019-09-16:
@coolsnowwolf 首先我之前的留言有些轻佻，我真诚道歉！

没仔细研究 SSR Plus+ 的流程，如果 80，443 最后是通过 iptables 转到 ss(r)-redir，那么改用 redsocks 肯定是可行的，也不会加大路由器负担。至于 QoS，Fast Path 我是真不清楚会不会影响。

[semigodking](https://github.com/semigodking/redsocks) 版本的 redsocks 有自动判定是否被墙的功能（前提是 DNS 不被污染），依据应该是一定时间内没有数据回复或者回复空数据，在一些网络状况下肯定会误判，但大多数情况还是挺有用，基本可以不用 GFWList，或者不用常更新这个列表。

下面是我现在用的表逻辑：

```
9     1010 61224 RETURN     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            match-set chnroute dst
10    2473  148K RETURN     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            match-set serverlist dst
12     255 15376 REDIRECT   tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            multiport dports 80,443,4070,9001,9418 match-set gfwlist dst redir ports 8092
13     458 28068 REDIRECT   tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            multiport dports 80,443,4070,9001,9418 redir ports 8902
```
先跳过国内和代理服务器地址，然后匹配 GFWList，最后跳转到 redsocks 的自动判定端口。

czb1994, 2019-09-17:
比较不明白为啥这个功能没有过滤域名某域名绝对不走工具出去

shikkoku, 2019-09-17:
> 你还是用gfw模式吧，打不开的就添加，我自己添了20多个。
> 我感觉这设计没问题
> 至于你说的全局模式下绕过局域网和大陆，再用bt，流量不走国外，我感觉可能性不大，也可能你测试时下载的bt没有能链接的国外地址吧，具体bt啥原理我不清楚，我觉得应该是我说的样子

电脑版所谓的全局模式只是设置了系统代理的软件会才会走全局，不设置系统代理的软件不会走SSR代理。路由客户端是真全局，所有软件流量都会经过路由转发SSR客户端处理，无关电脑上软件里的代理开关与否。

coolsnowwolf, 2019-09-17:
如果按照这个思路，只代理 80,443,4070,9001,9418  这几个特定端口，游戏党又乐意了，游戏的 UDP 互联端口是随机的。这种情况怎么解决呢？

如果 Readsocks 自己的判断出错了（某些域名被错误判断），明明可以直连的IP被走代理了，你打算如何解决？

rampageX, 2019-09-17:
1. UDP 的确不能处理，只能管 TCP ，上网浏览；
2. 它有时间设置，一段时间内会丢弃缓存的“它判断为被墙的域名”； 

```
autoproxy {
    no_quick_check_seconds = 3600; // Directly relay traffic to proxy if an IP
                                 // is found blocked in cache and it has been
                                 // added into cache no earlier than this
                                 // specified number of seconds. 3600
                                 // Set it to 0 if you do not want to perform
                                 // quick check when an IP is found in blocked
                                 // IP cache, thus the connection will be
                                 // redirected to proxy immediately.
    quick_connect_timeout = 3;   // Timeout value when performing quick
                                 // connection check if an IP is found blocked
                                 // in cache. 3
}

ipcache {
    // Configure IP cache
    cache_size = 4;   // Maximum number of IP's in 1K.
    stale_time = 86400; // Seconds to stale an IP in cache since it is added
                      // into cahce.
                      // Set it to 0 to disable cache stale.
    port_check = 1;   // Whether to distinguish port number in address
    cache_file = "/tmp/ipcache.txt"; // File used to store blocked IP's in cache.
    autosave_interval = 3600; // Interval for saving ip cache into file.
                              // Set it to 0 to disable autosave.
    // When autosave_interval and stale_time are both 0, IP cache behaves like
    // a static blacklist.
}
```

yasum2006, 2019-09-22:
我使用的方法：修改/etc/config/shadowsocksr文件，
config access_control
	option wan_bp_list '/etc/china_ssr.txt'
	option lan_ac_mode 'b'
	option router_proxy '1'

	option wan_bp_list '/etc/china_ssr.txt'
	option lan_ac_mode 'b'
	option router_proxy '0'
实际使用，当路由器挂载硬盘下载BT时，SSR Plus+会过滤BT流量

rufengsuixing, 2019-09-25:
关掉游戏模式udp还会重定向么

antidoom, 2019-09-27:
下载机专门整一个，实在要本地下载就用虚拟机里的下载软件被，设置该ip不走代理。没法全部兼顾啊。

linch0318, 2019-10-08:
bt难免遇到海外ip，我把下载机拉到lan黑名单里，不走代理，偶尔需要pc端下载的，也是先暂停任务，等最后一次性修改ssr规则再下载。
另外我觉得现在wan和lan的名单有点麻烦，只能一项项添加，希望可以直接一个列表进去，设备多了操作麻烦

linch0318, 2019-10-08:
gfw模式和绕过中国ip模式各有优缺点，从名字上就能想象得到，有些网站不在gfw里，要用绕过中国ip模式才能访问，有些网站能访问但速度很慢，这时候可以挂全局

closed 2019-10-18 by coolsnowwolf