coolsnowwolf/lede #1626: Sysupgrade not upgrading my router，无法通过Sysupgrade升级固件

yebr, 2019-09-15:
刷入OpenWrt R9.8.17 / LuCI Master (git-19.146.62144-fd6fdb2后，无法升级任何其它固件。
使用LUCI界面升级，不保留配置下升级固件，显示擦除中...但实际上只是路由器重启了一下。

使用命令行命令升级固件，提示：Commencing upgrade. Closing all shell sessions.后，也是实际上只是路由器重启了一下，无任何变化。

root@OpenWrt:/tmp/tmp#sysupgrade -n -v openwrt-mvebu-cortexa9-linksys_wrt1200ac-squashfs-sysupgrade.bin 
Commencing upgrade. Closing all shell sessions.

来自 192.168.1.1 的回复: 字节=32 时间<1ms TTL=64
来自 192.168.1.1 的回复: 字节=32 时间<1ms TTL=64
来自 192.168.1.1 的回复: 字节=32 时间<1ms TTL=64
来自 192.168.1.1 的回复: 字节=32 时间<1ms TTL=64
来自 192.168.1.1 的回复: 字节=32 时间<1ms TTL=64
来自 192.168.1.1 的回复: 字节=32 时间<1ms TTL=64
请求超时。
请求超时。
请求超时。
请求超时。
请求超时。
来自 192.168.1.1 的回复: 字节=32 时间<1ms TTL=64
来自 192.168.1.1 的回复: 字节=32 时间<1ms TTL=64
来自 192.168.1.1 的回复: 字节=32 时间<1ms TTL=64
来自 192.168.1.1 的回复: 字节=32 时间<1ms TTL=64
来自 192.168.1.1 的回复: 字节=32 时间<1ms TTL=64
来自 192.168.1.1 的回复: 字节=32 时间<1ms TTL=64
来自 192.168.1.1 的回复: 字节=32 时间<1ms TTL=64

yebr, 2019-09-15:
我的硬件是wrt1200ac

yebr, 2019-09-15:
root@OpenWrt:~# mount
/dev/root on /rom type squashfs (ro,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/ubi0_1 on /overlay type ubifs (rw,noatime,assert=read-only,ubi=0,vol=1)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
ubi1:syscfg on /tmp/syscfg type ubifs (rw,relatime,assert=read-only,ubi=1,vol=0)
tmpfs on /dev type tmpfs (rw,nosuid,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
/dev/ubi1_0 on /mnt/ubi1_0 type ubifs (rw,relatime,assert=read-only,ubi=1,vol=0)

root@OpenWrt:~# umount -a
umount: can't unmount /dev/pts: Resource busy
umount: can't unmount /dev: Resource busy
umount: can't unmount /: Resource busy
umount: can't unmount /tmp: Resource busy
umount: can't unmount /proc: Resource busy

root@OpenWrt:~# df
Filesystem           1K-blocks      Used Available Use% Mounted on
tmpfs                   255476       812    254664   0% /tmp
overlayfs:/overlay       14908       220     13892   2% /
tmpfs                      512         0       512   0% /dev

root@OpenWrt:/tmp/tmp# ls
openwrt-mvebu-cortexa9-linksys_wrt1200ac-squashfs-sysupgrade.bin
root@OpenWrt:/tmp/tmp# sysupgrade -F openwrt-mvebu-cortexa9-linksys_wrt1200ac-squashfs-sysupgrade.bin 
Saving config files...
Commencing upgrade. Closing all shell sessions.

还是不行！

yebr, 2019-09-15:
想切换回另一到分区的固件，提示：
ERROR: Unable to obtain firmware environment variable: boot_part. Unable to set firmware environment variable: bootcmd to run nandboot.

yebr, 2019-09-15:
手动快速关机开机3次，也无法进入另一个固件系统

yebr, 2019-09-15:
关机状态下，按住reset键10秒，也无法进入u-boot

yebr, 2019-09-15:
尝试过把安装了的ipk删除，删除了大部分应用后尝试升级，也不成功，我已经无从下手了

yebr, 2019-09-15:
root@OpenWrt:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00200000 00020000 "u-boot"
mtd1: 00040000 00020000 "u_env"
mtd2: 00040000 00020000 "s_env"
mtd3: 00100000 00020000 "devinfo"
mtd4: 02800000 00020000 "kernel1"
mtd5: 02200000 00020000 "ubi"
mtd6: 02800000 00020000 "kernel2"
mtd7: 02200000 00020000 "rootfs2"
mtd8: 02600000 00020000 "syscfg"
mtd9: 00680000 00020000 "unused_area"

yebr, 2019-09-15:
root@OpenWrt:/etc# fw_printenv 
Cannot parse config file '/etc/fw_env.config': No such file or directory

shikkoku, 2019-09-16:
什么平台，什么路由器？

shikkoku, 2019-09-16:
下次报告前先把路由器型号和平台标注清楚。

yebr, 2019-09-16:
解决方法：
echo /dev/mtd1 0x0 0x20000 0x40000 >> /etc/fw_env.config
fw_printenv显示信息为正常

closed 2019-09-16 by yebr