coolsnowwolf/lede #1628: lxc容器安装了一个debian，报错，希望路过大佬帮看解决下

cattypapa, 2019-09-16:
系统日志：
/usr/lib/lua/luci/dispatcher.lua:509: Failed to execute call dispatcher target for entry '/admin/services/lxc_configuration_get/debian'.
The called action terminated with an exception:
/usr/lib/lua/luci/controller/lxc.lua:136: attempt to index local 'f' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:509: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:127: in function </usr/lib/lua/luci/dispatcher.lua:126>

zhg-em, 2019-09-20:
用lxc-checkconfig命令查看缺什么。重新编译固件将缺的东西编译进去！

cattypapa, 2019-09-21:
> 用lxc-checkconfig命令查看缺什么。重新编译固件将缺的东西编译进去！

我发誓，我编译的时候把lxc的全部都选择上了，但是我用checkconfig还是提示我，netlink missiing，我的设备是N1。我现在很懵逼

slitilanber, 2019-10-11:
有没编译debootstap进固件？没有debootstrap,luci和命令行都安装不了。
另外，默认下，没有systemd的cgroup,凡采用systemd的发行版都不能完全正常使用

closed 2019-10-18 by coolsnowwolf