coolsnowwolf/lede #1637: x86编译报错，帮忙看看

ihuaer, 2019-09-18:
Building backport-include/backport/autoconf.h ... done.
make[9]: *** No rule to make target `/home/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.32-1/compat/main.o', needed by `/home/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.32-1/compat/compat.o'.  Stop.
make[8]: *** [/home/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.32-1/compat] Error 2
make[7]: *** [_module_/home/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.32-1] Error 2
make[6]: *** [modules] Error 2
make[5]: *** [modules] Error 2
make[4]: *** [modules] Error 2
make[4]: Leaving directory `/home/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.32-1'
make[3]: *** [/home/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.32-1/.built] Error 2
make[3]: Leaving directory `/home/openwrt/lede/package/kernel/mac80211'
time: package/kernel/mac80211/compile#6.05#3.30#723.64
make[2]: *** [package/kernel/mac80211/compile] Error 2
make[2]: Leaving directory `/home/openwrt/lede'
make[1]: *** [/home/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/openwrt/lede'
make: *** [world] Error 2
openwrt@ubuntu:~/lede$

wchqybs, 2019-09-19:
编译软件需要更新了sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint

ihuaer, 2019-09-19:
> 编译软件需要更新了sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core [...]

不行啊，还是报错。9.16号还好好的 ，更新之后就一直报错
Building backport-include/backport/autoconf.h ... done.
make[9]: *** No rule to make target `/home/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.32-1/compat/main.o', needed by `/home/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.32-1/compat/compat.o'.  Stop.
make[8]: *** [/home/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.32-1/compat] Error 2
make[7]: *** [_module_/home/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.32-1] Error 2
make[6]: *** [modules] Error 2
make[5]: *** [modules] Error 2
make[4]: *** [modules] Error 2
make[4]: Leaving directory `/home/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.32-1'
make[3]: *** [/home/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.32-1/.built] Error 2
make[3]: Leaving directory `/home/openwrt/lede/package/kernel/mac80211'
time: package/kernel/mac80211/compile#1.04#0.67#3.40
make[2]: *** [package/kernel/mac80211/compile] Error 2
make[2]: Leaving directory `/home/openwrt/lede'
make[1]: *** [/home/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/openwrt/lede'
make: *** [world] Error 2
openwrt@ubuntu:~/lede$

closed 2019-10-18 by coolsnowwolf