coolsnowwolf/lede #1648: v2ray若设置传输协议为ws，ws host和ws path为空，v2ray不启动

tanghannb, 2019-09-21:
ssr-plus中的v2ray，若设置传输协议为ws，ws host和ws path为空，v2ray不启动，在ws host里填上none，v2ray就启动并生效了。。。这样很麻烦，订阅的节点很多，要一个一个改，希望lean大看一看是什么问题

tanghannb, 2019-09-21:
在config/shadowsocksr文件里，在luci界面的ws host里填上none，最后三行变成	option transport 'ws'
	option security 'auto'
	option ws_host 'none'
在luci界面的ws host里不填信息，最后三行为	option transport 'ws'
	option tcp_guise 'none'
	option tls '0'
	option security 'auto'
是不是这里出问题了

copperion, 2019-09-21:
I have the same problem.

cdcvip, 2019-09-26:
感谢提供思路，我解决了

dongfangy2011, 2019-09-27:
不能正常运行的都可能是使用的移动网络，是移动的使用什么新的技术所致，其他网络还行

copperion, 2019-09-28:
我最近知道了，我用V2RayN导出的客户端配置里面streamSettings字段是这样的：
```
"streamSettings": {
        "network": "ws",
        "wsSettings": {
          "connectionReuse": true,
          "path": null,
          "headers": null
        }
}
```
而SSR-Plus生成出来的`/var/etc/v2-ssr-retcp.json`中相关字段是
```
"streamSettings": {
       "network": "ws",
       "tlsSettings": {
         "allowInsecure": false
       },
       "wsSettings": [
       ],
       "security": "none"
 }
```
相当于`"connectionReuse": true`这一条没有加上，但是SSR-Plus的UI上面又没有这一条设置，所以可能导致了启动失败的问题。

shikkoku, 2019-10-04:
我让机场站长加上了host和path能用了hhhh

closed 2019-10-18 by coolsnowwolf