coolsnowwolf/lede #1660: SSR插件绕过大陆模式下，如何屏蔽某些外网。

alechsu, 2019-09-24:
由于是公司环境，不想用户传播有关政治的信息，试着去除了gfwlist里面美国之音网站的几个域名，发现还是可以上，请教应该如何屏蔽。谢谢。

alechsu, 2019-09-24:
顺便说一下在gfwlist模式下把网站从列表删除是可以的做到的，在绕过中国大陆模式下应该怎么做？

shNanChen, 2019-09-24:
给你提供个思路，
利用dnsmasq的ipset功能，把不需要代理的域名解析至ipset，
然后iptables里匹配ipset，-J REJECT掉，
就是这样

magiclu, 2019-09-25:
为什么要屏蔽，说不定那群人说的是对的，毕竟现在xxx的可信度也很低，可以参考新疆

就算你屏蔽了一部分，也不能完全屏蔽，比如youtube上也有相关内容

alechsu, 2019-09-25:
> hosts不久行了，什么论子的新唐人的我都是这么屏蔽的

 hosts文件可以屏蔽顶级域名吗？比如屏蔽了test.com就等于屏蔽了xxx.test.com？还是说需要一个个屏蔽xxx.test.com, yyy.test.com, zzz.test.com这样？

hcym, 2019-09-25:
不行，怕麻烦就要用dnamasq

rufengsuixing, 2019-09-25:
不是有广告过滤的adblock plus ，黑名单走起啊

alechsu, 2019-09-26:
> 不是有广告过滤的adblock plus ，黑名单走起啊

我用了一个精简的插件里面没有这个。。。

alechsu, 2019-09-26:
> 不行，怕麻烦就要用dnamasq

谢谢，已用dnsmasq 配置文件加域名解决

closed 2019-09-26 by alechsu

fhbyljj, 2021-04-24:
> 为什么要屏蔽，说不定那群人说的是对的，毕竟现在xxx的可信度也很低，可以参考新疆
> 
> 就算你屏蔽了一部分，也不能完全屏蔽，某些youtube上也有相关内容

你说的都对