coolsnowwolf/lede #1665: 神奇的问题：DDNS域名无法访问IP却没有问题

greekstreet, 2019-09-25:
使用的Openwrt 18.06.4加上lean大的酸牛奶，正常情况下，通过WAN口的pppoe公网IP是可以无障碍访问https luci和ssh的，两个均改过端口，非常规443和22，且防火墙已经添加了开启端口的规则，最近发现奇葩问题如下：
备注：已确认ddns域名解析的是正确的WAN IP。

没有绑定WAN IP到ddns域名，酸牛奶（微图瑞）开启：外网通过WAN IP可以访问路由器https luci, ssh；
绑定WAN IP到ddns域名，酸牛奶(微图瑞)开启：外网无法通过ddns域名或者WAN IP访问路由器https luci, ssh，但是却可以ping通ddns域名或者WAN IP；
绑定WAN IP到ddns域名，酸牛奶(微图瑞)关闭：外网可以通过ddns域名或者WAN IP访问路由器https luci, ssh；

不知道有没有人遇到同样的问题。ddns通过ddns-scripts或者自己手动通过wget -qO-方式更新，都是一样的结果。求大神指点啊。

rufengsuixing, 2019-09-25:
你的国内ip列表下载失败

greekstreet, 2019-09-26:
> 你的国内ip列表下载失败

不应该啊，我用的gfw模式，按理来说应该跟国内ip列表没有关系啊。

slitilanber, 2019-09-26:
我试过的DDNS只有ALI正产，softether自带DDNS，afraid ping回来的ip都是墙外server的IP。其他没试过。koolshare的没这个问题。但不想用他的固件。自己不懂技术，不知道咋搞。

rufengsuixing, 2019-09-26:
自己看看防火墙规则吧，这东西别人帮不上忙，主要是对应端口input链是否开了，net 链的prerouting应该不出问题但是也看看

greekstreet, 2019-09-26:
定位到问题了，发现gfwlist里面竟然有好多ddns域名！去/usr/bin/ss-gfw里面把自己用到的ddns域名加到忽略列表就行了，这样每次更新gfwlist也不会把自己用的ddns域名放进去。关掉这个issue了。

closed 2019-09-26 by greekstreet

xzphot, 2023-08-01:
> 定位到问题了，发现gfwlist里面居然有好多ddns域名！去/usr/bin/ss-gfw里面把自己用到的ddns域名加到忽略列表就行了，这样每次更新gfwlist也不会把自己用的ddns 核心去。关掉这个问题了。

时隔4年，我也被这个问题困扰。感谢！搞定了。