coolsnowwolf/lede #1666: pppoe自动断开重拨无网络

Saxon-Sun, 2019-09-25:
日志在13:01分重新拨号后一直无网络 连接状态可以看到获取到了ip外网地址但无网络 必须手动去luci点wan连接后才能恢复网络    偶尔出现 不知道是什么原因  

固件版本 | OpenWrt R9.9.15 / LuCI Master (git-18.296.11678-2bbc1e3)
内核版本 | 4.19.72
[系统日志.txt](https://github.com/coolsnowwolf/lede/files/3651235/default.txt)
[内核日志.txt](https://github.com/coolsnowwolf/lede/files/3651265/default.txt)

coolsnowwolf, 2019-09-26:
看看是不是集成了第三方源的ipk 导致的

Saxon-Sun, 2019-09-26:
> 看看是不是集成了第三方源的ipk 导致的

用的软件包都是您的源码git下来直接勾选软件编译的 未添加其他第三方ipk  大雕 日志里面能看出什么问题吗？

closed 2019-09-29 by coolsnowwolf

fanck0605, 2020-05-14:
@Saxon-Sun 我也有相同的状况，表现为电信48小时把我踢下线后 PPPoE 会自动重连获取到 IP，但是无法连接网络。概率性，不是每次发生。

Saxon-Sun, 2020-05-14:
@fanck0605 我用笨方法解决了 
设置了个计划任务 0 5 * * * /sbin/ifup wan 每天5点钟 重启wan口临时解决了 
我感觉可能是48小时被踢了重拨有时候dhcp6获取不到ipv6的缘故  但是我没有取消v6的包编译去尝试是不是v6的缘故
反正到现在我都是设置定时重启wan口临时解决的  我坐标是电信0577