coolsnowwolf/lede #1672: DHCP/DNS 功能无法使用

aaron021512, 2019-09-25:
[image]

`/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute cbi dispatcher target for entry '/admin/network/dhcp'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:55: Failed to execute template 'cbi/map'.
A runtime error occured: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'cbi/tblsection'.
A runtime error occured: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'cbi/value'.
A runtime error occured: /usr/lib/lua/luci/model/cbi/admin_network/dhcp.lua:220: attempt to call field 'checkmac' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function `

如图，我从15升级到最新版本19，保留了原有配置升级的，我试了在'/etc/config/dhcp'把之前配置的静态配置删除，不报错了，但是点击新增，就又显示同样的错误了，请问有没有人碰到过一样的问题啊？

admi-229133, 2019-09-26:
如果是自己编译的把 LuCI 升级一下因为开始的那几版确实有BUG 后面的已经修复 还有一点就是刷机以后出厂设置一下可以减少各种莫名其妙的BUG  最新LUCI版本是 (git-19.268.50099-562a894)

coolsnowwolf, 2019-09-26:
已经修复

closed 2019-09-26 by coolsnowwolf