coolsnowwolf/lede #1683: 关于print_req_error: I/O error的问题

ok-day, 2019-09-29:
基本是默认的配置进行编译
安装好系统之后依然出现如下错误
[   98.187240] print_req_error: I/O error, dev loop0, sector 34040
[   98.193426] print_req_error: I/O error, dev loop0, sector 39312
[   98.193443] F2FS-fs (loop0): Issue discard(4255, 4255, 2) failed, ret: -5
[   98.199808] print_req_error: I/O error, dev loop0, sector 39352
[   98.207039] F2FS-fs (loop0): Issue discard(4914, 4914, 1) failed, ret: -5
[   98.213358] print_req_error: I/O error, dev loop0, sector 39376
[   98.220557] F2FS-fs (loop0): Issue discard(4919, 4919, 1) failed, ret: -5
[   98.234190] F2FS-fs (loop0): Issue discard(4922, 4922, 2) failed, ret: -5

ok-day, 2019-09-29:
源码clone 的是最新的
删除了lede目录 重新搞的
固件版本  OpenWrt R9.10.1 / LuCI Master (git-19.271.33176-b099749)
内核版本  4.19.75

ope087, 2019-09-29:
root分区不能大于128m

ok-day, 2019-09-29:
> root分区不能大于128m,或者使用ext4分区格式

root分区确实小于128 使用的是squashfs 谢谢解答 我去试试

ope087, 2019-09-29:
我之前设的256M,就经常看到报警,后来改成了120M,就没有再看到过了

ok-day, 2019-09-29:
> 我之前设的256M,就经常看到报警,后来改成了120M,就没有再看到过了

好的，我试试120，感谢。

closed 2019-09-29 by coolsnowwolf

thank243, 2019-12-19:
试下用

> fdisk /dev/sda

修复下？