coolsnowwolf/lede #1688: dnspod插件与代理冲突问题

MoeGrid, 2019-09-29:
`lede/package/lean/luci-app-dnspod/files/dnspod.sh` 文件第72行获取本机IP使用的是`whatismyip.akamai.com`，如果在代理开启绕过大陆IP地址或全局模式时，返回的是代理服务器的IP，导致DDNS解析错误。

解决方案：改为获取wan口IP地址
```
getIp() {
    ubus call network.interface.wan status | grep \"address\" | grep -oE '[0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}'
}
```

greennyreborn, 2019-09-30:
确实是个问题。
不过不是所有人都拿lede直接接外网的，大部分都是接在光猫或者主路由后面，你这个解法这种case下拿到的还是内网ip。
我想想有没有兼容这两种方式的办法

MoeGrid, 2019-09-30:
比如加个开关？让用户选择使用wan口还是在线获取

MoeGrid, 2019-09-30:
好像加开关也依然解决不了这个问题

greennyreborn, 2019-09-30:
> 好像加开关也依然解决不了这个问题

先从wan口取，再判断下是不是 public ip，如果不是就 curl api 取。
这种情况下 lede 做二级路由且开全局的话，还是会有问题的。

你是什么场景需要直接设置全局模式，而不能指定ip来开启全局模式呢

MoeGrid, 2019-09-30:
> > 好像加开关也依然解决不了这个问题
> 
> 先从wan口取，再判断下是不是 public ip，如果不是就 curl api 取。
> 这种情况下 lede 做二级路由且开全局的话，还是会有问题的。
> 
> 你是什么场景需要直接设置全局模式，而不能指定ip来开启全局模式呢

我是开的国内外IP分流，因为查IP的哪个地址是国外的，导致了这个问题，其实换成国内查IP的地址也能解决

Bnosora, 2019-10-01:
我这的场景是局域网里设备很多，除了负责下载的几个电脑以外都开白名单模式，这样可以避免有些域名访问较慢但没被墙所以不在gfwlist上的情况。算是把v2当作半个加速器用了。

Bnosora, 2019-10-01:
获取ip的网址换成国内的不知道能否解决问题？或者dnspod也有ddns的api，不需要用户发送公网ip过去。

greennyreborn, 2019-10-04:
https://github.com/coolsnowwolf/lede/pull/1705

closed 2019-10-09 by MoeGrid

Archie87, 2019-10-14:
我也遇到同样的问题，请问那串代码应该加再哪个位置？