coolsnowwolf/lede #1691: 讨论：zerotier 的forward rule 可能的安全性问题

rufengsuixing, 2019-09-30:
默认的forward是打开的
防火墙限制是通过iptables zone的对应规则实现的
zerotier.start里出现了以下代码
  iptables -I FORWARD -i $zt0 -j ACCEPT
  iptables -I FORWARD -o $zt0 -j ACCEPT
会导致iptables规则里
Chain FORWARD (policy DROP 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination
    0     0 DROP       all  --  *      *       0.0.0.0/0            0.0.0.0/0            match-set blockip dst
38047   24M ACCEPT     all  --  *      ztwfuebudm  0.0.0.0/0            0.0.0.0/0
36004   11M ACCEPT     all  --  ztwfuebudm *       0.0.0.0/0            0.0.0.0/0
 369K   23M forwarding_rule  all  --  *      *       0.0.0.0/0            0.0.0.0/0            /* !fw3: Custom forwarding rule chain */
 304K   19M FLOWOFFLOAD  all  --  *      *       0.0.0.0/0            0.0.0.0/0            /* !fw3: Traffic offloading */ ctstate RELATED,ESTABLISHED FLOWOFFLOAD
 304K   19M ACCEPT     all  --  *      *       0.0.0.0/0            0.0.0.0/0            ctstate RELATED,ESTABLISHED /* !fw3 */
65133 3306K zone_lan_forward  all  --  br-lan *       0.0.0.0/0            0.0.0.0/0            /* !fw3 */
  227 17982 zone_wan_forward  all  --  pppoe-wan *       0.0.0.0/0            0.0.0.0/0            /* !fw3 */
    0     0 reject     all  --  *      *       0.0.0.0/0            0.0.0.0/0            /* !fw3 */

这样如果数据包从wan口进入，访问zerotier的内网，是否会导致内网泄露？

rufengsuixing, 2019-09-30:
改成这样如何
  iptables -I forwarding_rule -i $zt0 -j ACCEPT
  iptables -I zone_lan_forward -o $zt0 -j ACCEPT

ZqinKing, 2019-11-12:
> 默认的forward是打开的
> 防火墙限制是通过iptables zone的对应规则实现的
> zerotier.start里出现了以下代码
> iptables -I FORWARD -i $zt0 -j ACCEPT
> iptables -I FORWARD -o $zt0 -j ACCEPT
> 会导致iptables规则里
> Chain FORWARD (policy DROP 0 packets, 0 [...]

Wan口进不来，防火墙上默认设置了Wan口进入的数据包丢弃。

rufengsuixing, 2019-11-20:
防火墙显示的有效的前提是你没有自己在防火墙定义好的链外写规则，在总链动手脚会导致你所看到的luci的显示变得不准，如果我在forward链第一个写一个accept，不关你luci防火墙怎么显示都是允许
另外我说的是转发，不是进入

closed 2019-12-24 by coolsnowwolf