coolsnowwolf/lede #1692: 关于编译mosquitto求教

madcat0513, 2019-09-30:
想编译luci-app-mosquitto进固件，但是make menuconfig里根本看不到这个选项，官方倒是能找到这个包的，不过依赖啥的太麻烦，装不上啊

stl88083365, 2019-09-30:
下载地址放上来看看。

madcat0513, 2019-10-01:
https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a15_neon-vfpv4/luci/luci-app-mosquitto_git-19.270.39227-355223e-1_all.ipk
https://downloads.openwrt.org/snapshots/packages/arm_cortex-a15_neon-vfpv4/luci/luci-app-mosquitto_git-19.271.67895-cf887ba-1_all.ipk

stl88083365, 2019-10-01:
L大的源码里有这个客户端啊  用  / 查看。

madcat0513, 2019-10-01:
没看懂，请问具体怎么操作？

stl88083365, 2019-10-01:
make menuconfig,然后 按键盘上的   “/”   按钮  输入你要搜索的

madcat0513, 2019-10-01:
~/openwrt/lean$ cat .config |grep mosquitto
# CONFIG_PACKAGE_lua-mosquitto is not set
# CONFIG_PACKAGE_libmosquitto-nossl is not set
# CONFIG_PACKAGE_libmosquitto-ssl is not set
# CONFIG_PACKAGE_mosquitto-client-nossl is not set
# CONFIG_PACKAGE_mosquitto-client-ssl is not set
# CONFIG_PACKAGE_mosquitto-nossl is not set
# CONFIG_PACKAGE_mosquitto-ssl is not set
还是没luci-app-mosquitto

closed 2019-10-18 by coolsnowwolf