coolsnowwolf/lede #1693: 新LUCI加速版本梯子有日志错误，还有更新按钮需要点2次，点击一次无效。

gpeditmsc1986, 2019-09-30:
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload: net.ipv4.tcp_congestion_control = cubic
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload: uci: Entry not found
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload: DNSMASQ change
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload: Warning: Section @zone[1] (wan) cannot resolve device of network 'wan'
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload: Warning: Section @zone[1] (wan) cannot resolve device of network 'wan6'
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload: Warning: Section @zone[1] (wan) has no device, network, subnet or extra options
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:  * Flushing IPv4 filter table
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:  * Flushing IPv4 nat table
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:  * Flushing IPv4 mangle table
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:  * Flushing IPv4 raw table
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:  * Flushing conntrack table ...
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:  * Populating IPv4 filter table
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:    * Rule 'Allow-DHCP-Renew'
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:    * Rule 'Allow-Ping'
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:    * Rule 'Allow-IGMP'
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:    * Rule 'Allow-IPSec-ESP'
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:    * Rule 'Allow-ISAKMP'
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:    * Forward 'lan' -> 'wan'
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:    * Zone 'lan'
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:    * Zone 'wan'
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:  * Populating IPv4 nat table
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:    * Zone 'lan'
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:    * Zone 'wan'
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload: Warning: wan will enable FULLCONE-NAT
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:  * Populating IPv4 mangle table
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:    * Zone 'lan'
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:    * Zone 'wan'
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:  * Populating IPv4 raw table
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:    * Zone 'lan'
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:      - Using automatic conntrack helper attachment
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:    * Zone 'wan'
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:  * Set tcp_ecn to off
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:  * Set tcp_syncookies to on
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:  * Set tcp_window_scaling to on
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:  * Running script '/etc/firewall.user'
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:  * Running script '/usr/share/miniupnpd/firewall.include'
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:  * Running script '/var/etc/shadowsocksr.include'
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:    ! Skipping due to path error: No such file or directory
Mon Sep 30 23:08:30 2019 daemon.notice procd: /etc/rc.d/S99autocore: net.core.rps_sock_flow_entries = 4096

stl88083365, 2019-09-30:
我这里没有这个报错。。看看是不是设置了哪里？

gpeditmsc1986, 2019-10-01:
> 我这里没有这个报错..看看是不是设置了哪里？

没有啊，基本上都是默认的设置，怕出问题，全新拉取编译的，还是有这个日志错误

stl88083365, 2019-10-01:
你是什么路由器？

gpeditmsc1986, 2019-10-01:
> 你是什​​么路由器？

X86的软路由，1900与3865的都是同样的错误日志

gpeditmsc1986, 2019-10-01:
[1900.config.txt](https://github.com/coolsnowwolf/lede/files/3675632/1900.config.txt)
这个是我的config文件

SouHong, 2019-10-16:
使用148内核编译的K3固件也同样有/etc/rc.d/S60flowoffload: ! Skipping due to path error: No such file or directory

closed 2019-10-18 by coolsnowwolf

gpeditmsc1986, 2019-10-18:
这个是我1900路由器，还不怎么会玩这个邮箱，我测试加速LUCI之后的版本都存在这个问题，加速之前这个是正常的，麻烦又空看看，感谢！！

Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.750329] system 00:07: [mem 0xe0000000-0xefffffff] could not be reserved
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.757288] system 00:07: [mem 0xfed01000-0xfed01fff] has been reserved
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.763912] system 00:07: [mem 0xfed03000-0xfed03fff] has been reserved
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.770528] system 00:07: [mem 0xfed04000-0xfed04fff] has been reserved
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.777148] system 00:07: [mem 0xfed0c000-0xfed0ffff] has been reserved
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.783758] system 00:07: [mem 0xfed08000-0xfed08fff] has been reserved
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.790378] system 00:07: [mem 0xfed1c000-0xfed1cfff] has been reserved
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.796988] system 00:07: [mem 0xfee00000-0xfeefffff] has been reserved
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.803607] system 00:07: [mem 0xfef00000-0xfeffffff] has been reserved
Fri Oct 18 22:45:39 2019 kern.debug kernel: [    1.810222] system 00:07: Plug and Play ACPI device, IDs PNP0c02 (active)
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.810623] pnp: PnP ACPI: found 8 devices
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.821490] clocksource: acpi_pm: mask: 0xffffff max_cycles: 0xffffff, max_idle_ns: 2085701024 ns
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.830398] clocksource: Switched to clocksource acpi_pm
Fri Oct 18 22:45:39 2019 kern.debug kernel: [    1.835759] pci 0000:00:1c.0: bridge window [mem 0x00100000-0x000fffff 64bit pref] to [bus 01] add_size 200000 add_align 100000
Fri Oct 18 22:45:39 2019 kern.debug kernel: [    1.835769] pci 0000:00:1c.1: bridge window [mem 0x00100000-0x000fffff 64bit pref] to [bus 02] add_size 200000 add_align 100000
Fri Oct 18 22:45:39 2019 kern.debug kernel: [    1.835779] pci 0000:00:1c.2: bridge window [mem 0x00100000-0x000fffff 64bit pref] to [bus 03] add_size 200000 add_align 100000
Fri Oct 18 22:45:39 2019 kern.debug kernel: [    1.835788] pci 0000:00:1c.3: bridge window [mem 0x00100000-0x000fffff 64bit pref] to [bus 04] add_size 200000 add_align 100000
Fri Oct 18 22:45:39 2019 kern.debug kernel: [    1.835797] pci 0000:00:1c.0: res[9]=[mem 0x00100000-0x000fffff 64bit pref] res_to_dev_res add_size 200000 min_align 100000
Fri Oct 18 22:45:39 2019 kern.debug kernel: [    1.835800] pci 0000:00:1c.0: res[9]=[mem 0x00100000-0x002fffff 64bit pref] res_to_dev_res add_size 200000 min_align 100000
Fri Oct 18 22:45:39 2019 kern.debug kernel: [    1.835803] pci 0000:00:1c.1: res[9]=[mem 0x00100000-0x000fffff 64bit pref] res_to_dev_res add_size 200000 min_align 100000
Fri Oct 18 22:45:39 2019 kern.debug kernel: [    1.835807] pci 0000:00:1c.1: res[9]=[mem 0x00100000-0x002fffff 64bit pref] res_to_dev_res add_size 200000 min_align 100000
Fri Oct 18 22:45:39 2019 kern.debug kernel: [    1.835810] pci 0000:00:1c.2: res[9]=[mem 0x00100000-0x000fffff 64bit pref] res_to_dev_res add_size 200000 min_align 100000
Fri Oct 18 22:45:39 2019 kern.debug kernel: [    1.835813] pci 0000:00:1c.2: res[9]=[mem 0x00100000-0x002fffff 64bit pref] res_to_dev_res add_size 200000 min_align 100000
Fri Oct 18 22:45:39 2019 kern.debug kernel: [    1.835817] pci 0000:00:1c.3: res[9]=[mem 0x00100000-0x000fffff 64bit pref] res_to_dev_res add_size 200000 min_align 100000
Fri Oct 18 22:45:39 2019 kern.debug kernel: [    1.835820] pci 0000:00:1c.3: res[9]=[mem 0x00100000-0x002fffff 64bit pref] res_to_dev_res add_size 200000 min_align 100000
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.835833] pci 0000:00:1c.0: BAR 9: no space for [mem size 0x00200000 64bit pref]
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.843416] pci 0000:00:1c.0: BAR 9: failed to assign [mem size 0x00200000 64bit pref]
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.851354] pci 0000:00:1c.1: BAR 9: no space for [mem size 0x00200000 64bit pref]
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.858921] pci 0000:00:1c.1: BAR 9: failed to assign [mem size 0x00200000 64bit pref]
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.866854] pci 0000:00:1c.2: BAR 9: no space for [mem size 0x00200000 64bit pref]
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.874434] pci 0000:00:1c.2: BAR 9: failed to assign [mem size 0x00200000 64bit pref]
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.882366] pci 0000:00:1c.3: BAR 9: no space for [mem size 0x00200000 64bit pref]
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.889933] pci 0000:00:1c.3: BAR 9: failed to assign [mem size 0x00200000 64bit pref]
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.897870] pci 0000:00:1c.3: BAR 9: no space for [mem size 0x00200000 64bit pref]
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.905448] pci 0000:00:1c.3: BAR 9: failed to assign [mem size 0x00200000 64bit pref]
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.913377] pci 0000:00:1c.2: BAR 9: no space for [mem size 0x00200000 64bit pref]
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.920960] pci 0000:00:1c.2: BAR 9: failed to assign [mem size 0x00200000 64bit pref]
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.928885] pci 0000:00:1c.1: BAR 9: no space for [mem size 0x00200000 64bit pref]
Fri Oct 18 22:45:39 2019 kern.info kernel: [    1.936469] pci 0000:00:1c.1: BAR 9: failed to assign [mem size 0x00200000 64bit pref]
[... cut ...]
Fri Oct 18 22:53:16 2019 kern.info /usr/bin/ssr-redir[12963]: running from root user
Fri Oct 18 22:53:16 2019 kern.info /usr/bin/ssr-redir[12965]: initializing ciphers... aes-128-cfb
Fri Oct 18 22:53:16 2019 kern.info /usr/bin/ssr-redir[12965]: tcp port reuse enabled
Fri Oct 18 22:53:16 2019 kern.info /usr/bin/ssr-redir[12965]: listening at 0.0.0.0:1234
Fri Oct 18 22:53:16 2019 kern.info /usr/bin/ssr-redir[12965]: running from root user
Fri Oct 18 22:53:16 2019 kern.info /usr/bin/ssr-redir[12967]: initializing c