coolsnowwolf/lede #1695: zerotier编译不过去

jovishow, 2019-10-01:
正常编译到zerotier后，报错如下：
make[3] -C feeds/packages/net/zerotier compile
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [world] Error 1

在menuconfig已关闭 network->VPN和luCI->Application后，执行make -j1 V=s仍然报错如下信息：
make[2]: Leaving directory `/Volumes/CompilerHD/lede'
make[1]: *** [/Volumes/CompilerHD/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/Volumes/CompilerHD/lede'
make: *** [world] Error 2

hanandjun, 2019-10-02:
兄台，还是老老实实在linux下跑吧，mac下我之前把所有用到这货的地方都在default列表里排除了，要是懂的话直接改改代码。或者你设置即使报错了还是继续编译，但是我没试过编译出来的能不能用

closed 2019-10-18 by coolsnowwolf

xiufeiwu, 2021-08-20:
我解决了，把Makefile里的
ifeq ($(OSTYPE),Darwin)
	include make-mac.mk
改为
ifeq ($(OSTYPE),Darwin)
	include make-linux.mk
竟然过了，我猜是用了homebrew安装了gnu的gcc，所以源码里解决macos编译的mk文件不适用；使用linux的就编译通过了