coolsnowwolf/lede #1697: 安装特定ipk包后luci出现错误

a843161264, 2019-10-01:
主机名 | OpenWrt
主机型号 | Phicomm K2P
架构 | MediaTek MT7621 ver:1 eco:3 x 4
固件版本 | OpenWrt R9.10.1 / LuCI Master (git-19.268.50099-562a894)
内核版本 | 4.14.146
错误描述：
原因：安装以下ipk包 odhcp6c odhcpd ip6tables luci-proto-ipv6luci
现象：
”luci-网络-接口“出现错误
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute arcombine dispatcher target for entry '/admin/network/network'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/network/proto_4x6.lua:61: attempt to call method 'register_error_code' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>
”luci-网络-交换机“出现错误
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute cbi dispatcher target for entry '/admin/network/vlan'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/network/proto_4x6.lua:61: attempt to call method 'register_error_code' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>

a843161264, 2019-10-01:
已找到冲突的软件包 luci-proto-ipv6luci ，卸载后正常。不知是否与大雕前几日更新了luci有关。

shikkoku, 2019-10-09:
luci你最好用lean的sdk编译或者是编译固件的时候集成，用openwrt官方源安装可能就会出现各种奇怪的问题。

closed 2019-10-10 by coolsnowwolf