coolsnowwolf/lede #1700: 请问如何还原被修改过的主页

Fr33raNg3r, 2019-10-02:
网上能找到一句话：
> \usr\lib\lua\luci\view\admin_status\index.htm 主页样式文件，温度显示等等
> X86架构固件的index，默认被大雕files强奸了，自己到源码里替换/lede/package/lean/autocore/files/index.htm

请问如果想换回官方的主页，是应该将哪个目录里的index.htm拷贝到哪个目录里，上面一句话有几种不同的理解。官方git里也没找到index.htm的文件。

hcym, 2019-10-02:
我的好像有，什么都没改的，很少在意，夏天j1900不过五十

ntlf9t, 2019-10-02:
/etc/index.htm

Fr33raNg3r, 2019-10-03:
> /etc/index.htm

大神，看不懂什么意思，你发的这个目录是源码里面还是哪里？找不到

ope087, 2019-10-04:
在luci-mod-admin-full这个包里,你可以用--force-reinstall强制安装下试试

shikkoku, 2019-10-04:
编译的时候去掉autocore你就可以得到“原版”

Fr33raNg3r, 2019-10-04:
> 编译的时候去掉autocore你就可以得到“原版”

感谢，我去试试。

Fr33raNg3r, 2019-10-04:
> 在luci-mod-admin-full这个包里,你可以用--force-reinstall强制安装下试试

> 编译的时候去掉autocore你就可以得到“原版”

我将源码里/feeds/luci/modules/luci-mod-admin-full/luasrc/view/admin_status/index.htm复制到package/lean/autocore/files/index.htm进行替换，在选中autocore的情况下，重新编译没有任何改变。
不知道是哪里的问题。

ope087, 2019-10-04:
> 
> 
> > 在luci-mod-admin-full这个包里,你可以用--force-reinstall强制安装下试试
> 
> > 编译的时候去掉autocore你就可以得到“原版”
> 
> 我将源码里/feeds/luci/modules/luci-mod-admin-full/luasrc/view/admin_status/index.htm复制到package/lean/autocore/files/index.htm进行替换，在选中autocore的情况下，重新编译没有任何改变。
> 不知道是哪里的问题。

清缓存或换浏览器试试,如果你备份的话可以对比下看看.

slitilanber, 2019-10-04:
1，修改autocore下index.htm，添加想要的信息：
如CPU温度，磁盘信息，smart信息，minidlna，upnp等。
2，make .../autocore/clean
3，重新编译固件

Fr33raNg3r, 2019-10-04:
> 1，修改autocore下index.htm，添加想要的信息：
> 如CPU温度，磁盘信息，smart信息，minidlna，upnp等。
> 2，make .../autocore/clean
> 3，重新编译固件

我不会前端，添加不来这些东西，不过还是谢谢了。

closed 2019-10-18 by coolsnowwolf