coolsnowwolf/lede #1704: 小娱c1/c5内存识别错误。

ggllioio, 2019-10-04:
RT，现在只能识别256MB内存。
实际应为512MB。
望修复。
谢谢。

shNanChen, 2019-10-05:
修改 `target/linux/ramips/dts/XiaoYu-C5.dts` 文件，
增加下面内容编译就可以识别512MB，
```
memory@0 {
	device_type = "memory";
	reg = <0x0 0x1c000000 0x20000000 0x4000000>;
};
```

ggllioio, 2019-10-05:
> 修改 `target/linux/ramips/dts/XiaoYu-C5.dts` 文件，
> 增加下面内容编译就可以识别512MB，
> 
> ```
> memory@0 {
> 	device_type = "memory";
> 	reg = <0x0 0x1c000000 0x20000000 0x4000000>;
> };
> ```

那就有点奇怪了，这段本来有的，在前些天被删掉了啊？
[https://github.com/coolsnowwolf/lede/commit/18a87273c0eae2847a6ca09f493d7fa9bb42add9](url)
19-21行。

shNanChen, 2019-10-05:
看上去是同步了上游官方原版的master树的xiaoyu适配，`https://github.com/openwrt/openwrt/blob/master/target/linux/ramips/dts/mt7621_xiaoyu_xy-c5.dts`，应该是master主分支是有内存自动识别的，不需要这段代码，
但是原版19.07分支是不能自动识别出512MB内存的，
具体这边lede是同步上游到哪了就不清楚了

fanvinga, 2019-10-09:
@ggllioio @shNanChen 在新的同步里面, https://github.com/coolsnowwolf/lede/commit/1aaf35221a9c02b415de44ec5ce5bb6525d33b09 这个commit已经有自动识别内存啦

closed 2019-10-11 by ggllioio