coolsnowwolf/lede #1708: kcptun客户端每次开机都不能正常启动，未运行

angrysad, 2019-10-05:
用的lean大的lede，ssrplus里ssr设置了kcptun的本地ip以及本地端口，kcp客户端添加好了服务器，可以正常使用，但是每次重启路由或者重新开机后，kcp客户端都运行不起来，一定要手动kcp客户端服务器选择禁用，再选择服务器才能启动客户端，显示运行中，希望有知道的大佬帮忙解决下。

angrysad, 2019-10-05:
Sat Oct  5 22:38:56 2019 daemon.err kcptun: Config validate failed.
Sat Oct  5 22:38:56 2019 daemon.notice procd: /etc/rc.d/S99kcptun: kcptun.general.log_folder=/var/log/kcptun validates as directory with false
Sat Oct  5 22:38:56 2019 daemon.notice procd: /etc/rc.d/S99kcptun: kcptun.general.enable_logging=1 validates as bool with true
Sat Oct  5 22:38:56 2019 daemon.notice procd: /etc/rc.d/S99kcptun: kcptun.general.daemon_user=root validates as string with true
Sat Oct  5 22:38:56 2019 daemon.notice procd: /etc/rc.d/S99kcptun: kcptun.general.client_file=/usr/bin/kcptun-client validates as string with true
Sat Oct  5 22:38:56 2019 daemon.notice procd: /etc/rc.d/S99kcptun: kcptun.general.server=cfg024a8f validates as uciname with true

每次重启或者开机，kcptun都报这个错

angrysad, 2019-10-05:
困扰1个多月了，希望各位大神帮忙指导一下，或者教我用ssrplus里怎么勾选kcp正常使用也行，谢谢各位了~~~

stl88083365, 2019-10-05:
升级最新版的KT试试呢？

angrysad, 2019-10-05:
最新版了，9-10的

stl88083365, 2019-10-05:
S99kcptun 这个文件发上来看下，还有配置文件

angrysad, 2019-10-05:
配置文件是指ssr的？

angrysad, 2019-10-05:
START=1                                                 USE_PROCD=1                                                                                                     KCPTUN=kcptun                                           CONFIG_FOLDER=/var/etc/$KCPTUN                                                                                  if [ -r /usr/share/libubox/jshn.sh ]; then                      . /usr/share/libubox/jshn.sh                    elif [ -r /lib/functions/jshn.sh ]; then                        . /lib/functions/jshn.sh                        else                                                            logger -p daemon.err -t "$KCPTUN" \                             "Package required: jshn."                       exit 1                                          fi                                                                                                              _log() {                                                        local level="$1"                                        local msg="$2"                                                                                                  logger -p "daemon.${level}" -t "$KCPTUN" "$msg" }                                                                                                               gen_client_config_file() {                                      local config_file="$1"                                                                                          json_init                                       - S99kcptun 1/238 0%

angrysad, 2019-10-05:
请问你有微信qq么？可以加你请教下么？不过得早上开电脑

stl88083365, 2019-10-05:
兄弟，你进群把。在群里说。大雕的群

angrysad, 2019-10-05:
在里面了，没人回答我呢……https://t.me/openwrtx86，是这个么？

stl88083365, 2019-10-05:
没人啊，你说句话

stl88083365, 2019-10-05:
https://t.me/opbypd

angrysad, 2019-10-05:
刚加说话了

closed 2019-10-17 by coolsnowwolf