coolsnowwolf/lede #1716: 编译报错

sunqm2015, 2019-10-08:
编译中出现报错信息，google cloud上运行

make[3]: Entering directory `/home/lede/lede/package/lean/openwrt-fullconenat'
touch /home/ledelede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/.prepared_2fd41f3a79b3612322b835d388e1b5d1_6664517399ebbbc92a37c5bb081b5c53_check
. /home/lede/lede/include/shell.sh; xzcat /home/lede/lede/dl/fullconenat-2018-12-15-d4daedd0.tar.xz | tar -C /home/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/.. -xf -
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
make[3]: *** [/home/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/.prepared_2fd41f3a79b3612322b835d388e1b5d1_6664517399ebbbc92a37c5bb081b5c53] Error 2
make[3]: Leaving directory `/home/lede/lede/package/lean/openwrt-fullconenat'
time: package/lean/openwrt-fullconenat/compile#0.04#0.01#0.38
make[2]: *** [package/lean/openwrt-fullconenat/compile] Error 2
make[2]: Leaving directory `/home/lede/lede'
make[1]: *** [/home/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/lede/lede'
make: *** [world] Error 2

请大佬看看，谢谢！

peeweep, 2019-10-08:
#1337

sunqm2015, 2019-10-08:
已经是Google cloud，不存在墙

peeweep, 2019-10-08:
删除 对应的 tar 文件重新下载 /home/lede/lede/dl/fullconenat-2018-12-15-d4daedd0.tar.xz

sunqm2015, 2019-10-08:
我再试试

closed 2019-10-10 by coolsnowwolf