coolsnowwolf/lede #1719: 更新内核后4.9编译报错

libinanshu, 2019-10-08:
1 out of 7 hunks FAILED -- saving rejects to file drivers/mtd/chips/cfi_cmdset_0002.c.rej
Patch failed!  Please fix /home/xl/lede/target/linux/ar71xx/patches-4.9/403-mtd_fix_cfi_cmdset_0002_status_check.patch!
Makefile:24: recipe for target '/home/xl/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.9.196/.prepared_1f9ca53250deba9e9f5894a4a1aab46f' failed
make[4]: *** [/home/xl/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.9.196/.prepared_1f9ca53250deba9e9f5894a4a1aab46f] Error 1
make[4]: Leaving directory '/home/xl/lede/target/linux/ar71xx'
Makefile:13: recipe for target 'compile' failed
make[3]: *** [compile] Error 2
make[3]: Leaving directory '/home/xl/lede/target/linux'
time: target/linux/compile#1.71#10.94#9.04
target/Makefile:23: recipe for target 'target/linux/compile' failed
make[2]: *** [target/linux/compile] Error 2
make[2]: Leaving directory '/home/xl/lede'
target/Makefile:18: recipe for target '/home/xl/lede/staging_dir/target-mips_24kc_musl/stamp/.target_compile' failed
make[1]: *** [/home/xl/lede/staging_dir/target-mips_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/xl/lede'
/home/xl/lede/include/toplevel.mk:216: recipe for target 'world' failed

nature2608, 2019-10-08:
一样报错 ar71xx

ope087, 2019-10-09:
文件打补丁的时候出错了,要么改回内核,要么修改补丁文件

closed 2019-10-10 by coolsnowwolf