coolsnowwolf/lede #1722: SSR plus最新版节点管理页面出错

lifetyper, 2019-10-09:
刚更新的最新版本，无论是编辑节点还是添加节点，都直接返回下面的错误：

/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute arcombine dispatcher target for entry '/admin/services/shadowsocksr/servers/cfg0a4a8f'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/nsection'.
A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/ucisection'.
A runtime error occured: /usr/lib/lua/luci/template.lua:74: Failed to load template 'shadowsocksr/ssrurl'.
Error while parsing template '/usr/lib/lua/luci/view/shadowsocksr/ssrurl.htm':
Syntax error in /usr/lib/lua/luci/view/shadowsocksr/ssrurl.htm:1: unfinished string near '"'
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function

lifetyper, 2019-10-09:
我编译的时候直接用files保留了旧版本的配置，不知道有没有影响。

MoeGrid, 2019-10-09:
我也是遇到了这个问题

lifetyper, 2019-10-09:
修的真tm快。。。。

closed 2019-10-09 by lifetyper