coolsnowwolf/lede #1734: Docker剩余空间格式化注意事项

RealKiro, 2019-10-11:
[image]
按照无脑配置教程默认格式化为sda3(源码)

网友说通常UEFI和MBR分区是不一样的，默认MBR通常是sda3，而UEFI通常是sda5；
我用的UEFI，这就需要[手动进行1～2步](https://github.com/RealKiro/lede/releases/tag/20191011)
[image]

---
❓有个小问题：目前你们升级固件Docker相关配置会丢失吗？

ope087, 2019-10-12:
plus插件也一直是升级丢配置的

coolsnowwolf, 2019-10-12:
只有 MBR 能保留配置，不推荐用 UEFI

ope087, 2019-10-12:
> 
> 
> plus插件也一直是升级丢配置的

好吧,没看清你说的升级估计丢配置,我是单升plus会丢,至少gfw列表会丢

lihaixin, 2019-10-12:
docker top CONTAINER 功能不能使用,提示没有-e功能

RealKiro, 2019-10-12:
> docker top CONTAINER 功能不能使用,提示没有-e功能

有问题优先考虑重启能不能解决

hcym, 2019-10-12:
蜗牛只能ufi，难受

lihaixin, 2019-10-12:
> > docker top CONTAINER 功能不能使用,提示没有-e功能
> 
> 有问题优先考虑重启路由能不能解决
编译好几次,也重启了,是不是openwrt不支持docker top 列出容器进程功能? 你们没有类似情况?
oot@openwrt:~# docker top ui
Error response from daemon: ps: ps: unrecognized option: e

coolsnowwolf, 2019-10-13:
ps -e 是因为内置的 busybox 不支持这个参数，安装 procps-ng-ps  

opkg update 后 opkg install procps-ng-ps --force-overwrite 即可解决

lihaixin, 2019-10-13:
使用opkg强行安装和直接在源码里编译安装procps-ng-ps 后,可以使用docker top 命令,但是会和ShadowSocksR Plus+ 里面的客户端状态显示冲突,安装procps-ng-ps  显示ShadowsocksR Plus+ 未运行,实际已经运行了,卸载procps-ng-ps后,就能正确显示状态

coolsnowwolf, 2019-10-14:
> 使用opkg强行安装和直接在源码里编译安装procps-ng-ps 后,可以使用docker top 命令,但是会和ShadowSocksR Plus+ 里面的客户端状态显示冲突,安装procps-ng-ps 显示ShadowsocksR Plus+ 未运行,实际已经运行了,卸载procps-ng-ps后,就能正确显示状态

已经更新代码兼容了

lihaixin, 2019-10-14:
thanks coolsnowwolf ! 已测试docker top ok,另外ttyd 管理界面,在https管理界面下不能正确调用,可否修复兼容https下嵌套

zhg-em, 2019-10-15:
修改后的固件基本正常，但打开docker CE 容器，显示Docker CE 未运行!
事实上应该该是运行的。

lihaixin, 2019-10-15:
终端查看  pgrep /usr/bin/dockerd 看是否返回进程ID,我这测试没问题

zhg-em, 2019-10-15:
> 终端查看 pgrep /usr/bin/dockerd 看是否返回进程ID,我这测试没问题

dockerd 是正常的，就是界面显示不正常！

lihaixin, 2019-10-15:
我使用最新的源码，需要把procps-ng-ps 编译进去才能显示，咋你的情况和我完全相反

closed 2019-10-18 by coolsnowwolf