coolsnowwolf/lede #1747: 在VULTR上的Ubuntu 16.04 x64编译10.10出错，求解决

realkobe, 2019-10-15:
( cd /home/wwz/lede/build_dir/hostpkg/go-1.12.1/src ; GOROOT_FINAL=/home/wwz/lede/staging_dir/hostpkg/lib/go-cross  GOROOT_BOOTSTRAP=/home/wwz/lede/build_dir/hostpkg/go-1.12.1/.go_bootstrap CC=gcc CXX=g++  bash make.bash --no-banner ; )
Building Go cmd/dist using /home/wwz/lede/build_dir/hostpkg/go-1.12.1/.go_bootstrap.
Building Go toolchain1 using /home/wwz/lede/build_dir/hostpkg/go-1.12.1/.go_bootstrap.
go build bootstrap/cmd/compile/internal/ssa: /home/wwz/lede/build_dir/hostpkg/go-1.12.1/.go_bootstrap/pkg/tool/linux_amd64/6g: signal: killed
go tool dist: FAILED: /home/wwz/lede/build_dir/hostpkg/go-1.12.1/.go_bootstrap/bin/go install -gcflags=-l -tags=math_big_pure_go compiler_bootstrap bootstrap/cmd/...: exit status 1
Makefile:271: recipe for target '/home/wwz/lede/build_dir/hostpkg/go-1.12.1/.built' failed
make[3]: *** [/home/wwz/lede/build_dir/hostpkg/go-1.12.1/.built] Error 2
make[3]: Leaving directory '/home/wwz/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#90.35#11.75#110.44
package/Makefile:107: recipe for target 'package/feeds/packages/golang/host/compile' failed
make[2]: *** [package/feeds/packages/golang/host/compile] Error 2
make[2]: Leaving directory '/home/wwz/lede'
package/Makefile:103: recipe for target '/home/wwz/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/wwz/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/wwz/lede'
/home/wwz/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

aliyuchang33, 2019-10-15:
```
make[1]: *** [tools/Makefile:153: /home/hork/build/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.tools_compile_yynyyyyynyyyyynyynnnyyyynyyyyyyyyyyyyyyynnyynynnyyynnyy] Error 2
make[1]: Leaving directory '/home/hork/build/lede'
make: *** [/home/hork/build/lede/include/toplevel.mk:218: world] Error 2
```
类似错误，没有头绪

ope087, 2019-10-15:
被杀了,内存不够?

aliyuchang33, 2019-10-15:
不会吧，24G内存呢

aliyuchang33, 2019-10-15:
gmake映射错了，软链接了一下make就好了

aliyuchang33, 2019-10-15:
我的是gmake映射不对，导致某些编译出问题，可以试试删除gmake然后软链接make

ope087, 2019-10-16:
> 
> 
> 不会吧，24G内存呢

go build bootstrap/cmd/compile/internal/ssa: /home/wwz/lede/build_dir/hostpkg/go-1.12.1/.go_bootstrap/pkg/tool/linux_amd64/6g: **signal: killed**
他的错误是这一行,单看最后一行是没有意义的

realkobe, 2019-10-17:
感谢各位，是内存问题，加大内存就解决了！

closed 2019-10-17 by coolsnowwolf