coolsnowwolf/lede #1752: 编译最新的系统没有ipv6支持

ShallCanyon, 2019-10-15:
router: Newifi3-d2
version: OpenWrt R9.10.10
想用校园网的ipv6，已经在menuconfig里添加了odhcpd和odhcp6c并编译完成，也编辑了etc/config/dhcp，#627 里提到了安装这两个包就可以启用ipv6但我的路由依旧获取不到v6地址
在路由器里：
```
root@OpenWrt:~# ls /lib/netifd/proto
dhcp.sh     dhcpv6.sh   minieap.sh  ppp.sh
```
```
root@OpenWrt:~# ifstatus wan6
{
  "up": false,
  "pending": true,
  "available": true,
  "autostart": true,
  "dynamic": false,
  "proto": "dhcpv6",
  "device": "eth0.2",
  "data": {
    
  }
}
```
之前刷官方openwrt并且编辑etc/config/dhcp后是可以正常启用ipv6的
有大佬知道怎么解决么？

PoiYuudachi, 2019-10-15:
Extra packages里选上ipv6helper就可以了

ShallCanyon, 2019-10-16:
> Extra packages里选上ipv6helper就可以了

 实测不行，还是没有v6地址

viagram, 2019-10-16:
我的K3也是一样，求解！

closed 2019-10-18 by coolsnowwolf

SiuKam, 2020-01-16:
我这边编译之后也是这样，已经勾选ipv6helper，情况相同