coolsnowwolf/lede #1757: 树莓派3 编译ssr+插件失败

blackjelly, 2019-10-16:
配置文件：
```
make[6]: Entering directory '/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/shadowsocksr-libev/shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543/src'
  CC       libshadowsocks_libev_la-utils.lo
  CC       libshadowsocks_libev_la-jconf.lo
  CC       libshadowsocks_libev_la-json.lo
  CC       libshadowsocks_libev_la-encrypt.lo
  CC       libshadowsocks_libev_la-udprelay.lo
  CC       libshadowsocks_libev_la-cache.lo
  CC       libshadowsocks_libev_la-acl.lo
  CC       libshadowsocks_libev_la-netutils.lo
  CC       libshadowsocks_libev_la-local.lo
local.c: In function 'server_recv_cb':
local.c:721:45: error: argument to 'sizeof' in 'strncpy' call is the same expression as the source; did you mean to use the size of the destination? [-Werror=sizeof-pointer-memaccess]
                     strncpy(host, ip, sizeof(ip));
                                             ^
cc1: all warnings being treated as errors
Makefile:791: recipe for target 'libshadowsocks_libev_la-local.lo' failed
make[6]: *** [libshadowsocks_libev_la-local.lo] Error 1
make[6]: Leaving directory '/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/shadowsocksr-libev/shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543/src'
Makefile:479: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/shadowsocksr-libev/shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543'
Makefile:388: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/shadowsocksr-libev/shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543'
Makefile:64: recipe for target '/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/shadowsocksr-libev//shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543/.built' failed
make[3]: *** [/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/shadowsocksr-libev//shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543/.built] Error 2
make[3]: Leaving directory '/root/openwrt/package/lean/shadowsocksr-libev'
time: package/lean/shadowsocksr-libev/compile#63.38#10.65#90.14
package/Makefile:111: recipe for target 'package/lean/shadowsocksr-libev/compile' failed
make[2]: *** [package/lean/shadowsocksr-libev/compile] Error 2
make[2]: Leaving directory '/root/openwrt'
package/Makefile:107: recipe for target '/root/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile' failed
make[1]: *** [/root/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/openwrt'
/root/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```
错误日志：
```
make[6]: Entering directory '/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/shadowsocksr-libev/shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543/src'
  CC       libshadowsocks_libev_la-utils.lo
  CC       libshadowsocks_libev_la-jconf.lo
  CC       libshadowsocks_libev_la-json.lo
  CC       libshadowsocks_libev_la-encrypt.lo
  CC       libshadowsocks_libev_la-udprelay.lo
  CC       libshadowsocks_libev_la-cache.lo
  CC       libshadowsocks_libev_la-acl.lo
  CC       libshadowsocks_libev_la-netutils.lo
  CC       libshadowsocks_libev_la-local.lo
local.c: In function 'server_recv_cb':
local.c:721:45: error: argument to 'sizeof' in 'strncpy' call is the same expression as the source; did you mean to use the size of the destination? [-Werror=sizeof-pointer-memaccess]
                     strncpy(host, ip, sizeof(ip));
                                             ^
cc1: all warnings being treated as errors
Makefile:791: recipe for target 'libshadowsocks_libev_la-local.lo' failed
make[6]: *** [libshadowsocks_libev_la-local.lo] Error 1
make[6]: Leaving directory '/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/shadowsocksr-libev/shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543/src'
Makefile:479: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/shadowsocksr-libev/shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543'
Makefile:388: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/shadowsocksr-libev/shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543'
Makefile:64: recipe for target '/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/shadowsocksr-libev//shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543/.built' failed
make[3]: *** [/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/shadowsocksr-libev//shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543/.built] Error 2
make[3]: Leaving directory '/root/openwrt/package/lean/shadowsocksr-libev'
time: package/lean/shadowsocksr-libev/compile#63.38#10.65#90.14
package/Makefile:111: recipe for target 'package/lean/shadowsocksr-libev/compile' failed
make[2]: *** [package/lean/shadowsocksr-libev/compile] Error 2
make[2]: Leaving directory '/root/openwrt'
package/Makefile:107: recipe for target '/root/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile' failed
make[1]: *** [/root/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/openwrt'
/root/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```

折腾好几天了，始终不过。

hcym, 2019-10-16:
肯定没问题，这个刚刚编译给n1用的，就是树3配置

blackjelly, 2019-10-16:
> 肯定没问题，这个刚刚编译给n1用的，就是树3配置

你肯定是用的L大的源码编译的，我是移植到openwrt官方的，用L大的源码编译后没有WiFi。

hcym, 2019-10-16:
哦，这个移植问题多多，我是放弃了，要么官方加其他ss插件，性能也不错，就是没有lean的方便

blackjelly, 2019-10-16:
> 哦，这个移植问题多多，我是放弃了，要么官方加其他ss插件，性能也不错，就是没有lean的方便

主要是图个方便而已，之前一直可能编译，但是最近几天就不行了，用也不急着用，就是搞不明白为什么

redchenjs, 2019-10-16:
@blackjelly 很简单，解压把patch丢patches文件夹里就行了
[999-Fix-Werror-sizeof-pointer-memaccess.patch.zip](https://github.com/coolsnowwolf/lede/files/3734488/999-Fix-Werror-sizeof-pointer-memaccess.patch.zip)

coolsnowwolf, 2019-10-16:
did you mean to use the size of the destination? [-Werror=sizeof-pointer-memaccess]
                     strncpy(host, ip, sizeof(ip));
 新版 GCC 编译器在 strncpy 禁止使用  sizeof 这个参数了

closed 2019-10-17 by coolsnowwolf