coolsnowwolf/lede #1758: 4.14.148/149,更新v2ray订阅链接问题

tengyuankoo, 2019-10-16:
服务器更新订阅全部空白，进不去客户端和高级设置选项卡
[image]

tengyuankoo, 2019-10-16:
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute function dispatcher target for entry '/admin/services/shadowsocksr'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute cbi dispatcher target for entry '/admin/services/shadowsocksr/client'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/shadowsocksr/client.lua:18: bad argument #1 to 'upper' (string expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>

coolsnowwolf, 2019-10-16:
不要保留配置升级

closed 2019-10-16 by coolsnowwolf