coolsnowwolf/lede #1759: 二次编译rnuc报错

bin20088, 2019-10-16:
二次编译只要不执行make clean就会在rnuc处报错，执行了make clean再编译就没问题，但是时间太长，不知道是什么原因。报错如下
make[3]: Entering directory '/home/rsa-key-20180509ooxxbbbbb/lede/feeds/packages/lang/golang/golang'
make[3]: Leaving directory '/home/rsa-key-20180509ooxxbbbbb/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#0.13#0.20#1.86
make[3]: Entering directory '/home/rsa-key-20180509ooxxbbbbb/lede/feeds/packages/utils/runc'
mkdir -p /home/rsa-key-20180509ooxxbbbbb/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/rsa-key-20180509ooxxbbbbb/lede/tmp/.root-copy.flock -c 'cp -fpR /home/rsa-key-20180509ooxxbbbbb/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8/.pkgdir/runc/. /home/rsa-key-20180509ooxxbbbbb/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/rsa-key-20180509ooxxbbbbb/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.runc_installed
mkdir -p /home/rsa-key-20180509ooxxbbbbb/lede/bin/targets/x86/64/packages /home/rsa-key-20180509ooxxbbbbb/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8/ipkg-x86_64/runc/CONTROL /home/rsa-key-20180509ooxxbbbbb/lede/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/rsa-key-20180509ooxxbbbbb/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8/ipkg-x86_64/runc/usr/sbin/
install -m0755 /home/rsa-key-20180509ooxxbbbbb/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8/.go_work/build/src/github.com/opencontainers/runc/runc /home/rsa-key-20180509ooxxbbbbb/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8/ipkg-x86_64/runc/usr/sbin/
install: cannot stat '/home/rsa-key-20180509ooxxbbbbb/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8/.go_work/build/src/github.com/opencontainers/runc/runc': No such file or directory
Makefile:88: recipe for target '/home/rsa-key-20180509ooxxbbbbb/lede/bin/packages/x86_64/packages/runc_1.0.0-rc8-1_x86_64.ipk' failed
make[3]: *** [/home/rsa-key-20180509ooxxbbbbb/lede/bin/packages/x86_64/packages/runc_1.0.0-rc8-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/rsa-key-20180509ooxxbbbbb/lede/feeds/packages/utils/runc'

closed 2019-10-16 by coolsnowwolf

whlsxl, 2020-02-19:
我还是有这个问题，二次编译不clean，就在这出错。

sqliuchang, 2020-02-19:
我也是 二次编译必须clean 否则runc报错