coolsnowwolf/lede #1760: Docker容器默认bridge模式无法访问外部网络

CirnoSalt, 2019-10-16:
硬件：J1900+PCIE双口网卡
固件版本：OpenWrt R9.10.10
网络配置：eth1-WAN/PPPoE、eth0-LAN
故障描述：U盘全新写入固件，安装docker过程一切顺利，WEB管理界面正常访问，
建立ubuntu容器后发现无法正常更新软件源，
[image]
由于Ubuntu镜像不包含ping工具，遂更换为centos容器下进行ping测试，
经过测试后发现容器内可正常ping通局域网LAN网段和docker0网桥的172网关地址，
但所有外网IP均无法ping通，DNS解析却正常。
[image]
（192.168.2.1为LAN地址）
随后检查了docker0网桥的状况，并未发现异常
[image]
已试过多次重启、建立iptables等方法均无效，
个人猜测为docker与多网口/op防火墙存在兼容性问题。

KFERMercer, 2019-10-16:
op的防火墙确实有问题。比如重启不会自动应用自定义的iptables rule

coolsnowwolf, 2019-10-16:
[image]

coolsnowwolf, 2019-10-16:
并不存在这个问题

KFERMercer, 2019-10-16:
> 并不存在这个问题

这个问题确实存在，因为我同样同样出现了这个问题：在首次拉取运行portainer时进入web界面会提示不能访问外部网络，之后运行别的容器 (比如人人客户端) 也会登录失败。手动重启防火墙后暂时恢复正常，但是重启路由器就会恢复到不能访问的状态。内核版本是4.19, 10月16日晚编译, 事前已重新configure。

KFERMercer, 2019-10-16:
> > 并不存在这个问题
> 
> 这个问题确实存在，因为我同样同样出现了这个问题：在首次拉取运行portainer时进入web界面会提示不能访问外部网络，之后运行别的容器 (比如人人客户端) 也会登录失败。手动重启防火墙后暂时恢复正常，但是重启路由器就会恢复到不能访问的状态。内核版本是4.19, 10月16日晚编译, 事前已重新configure。

在更早前的版本 (尚未添加docker) 中，防火墙就存在启动系统不会自动应用自定义的iptables rule的问题。这迫使我为了使规则生效，只能在启动脚本中手动添加重启防火墙的命令。我也测试过官方19.07分支代码编译的镜像，也会出现和上述同样的问题。这是op防火墙的bug吗？(测试内核: 4.19, 4.14)

CirnoSalt, 2019-10-16:
> > 并不存在这个问题
> 
> 这个问题确实存在，因为我同样同样出现了这个问题：在首次拉取运行portainer时进入web界面会提示不能访问外部网络，之后运行别的容器 (比如人人客户端) 也会登录失败。手动重启防火墙后暂时恢复正常，但是重启路由器就会恢复到不能访问的状态。内核版本是4.19, 10月16日晚编译, 事前已重新configure。

看来这个bug还有随机性的，这就很骚了

coolsnowwolf, 2019-10-16:
[image]

coolsnowwolf, 2019-10-16:
并不存在随机的问题，请更新你们的代码， 这个在线音乐播放器 in Docker 可以完整运行就是实例。

请 rm -rf build_dir ， rm .config, rm -rf tmp 后，更新 feeds 后编译

KFERMercer, 2019-10-16:
> > > 并不存在这个问题
> > 
> > 
> > 这个问题确实存在，因为我同样同样出现了这个问题：在首次拉取运行portainer时进入web界面会提示不能访问外部网络，之后运行别的容器 (比如人人客户端) 也会登录失败。手动重启防火墙后暂时恢复正常，但是重启路由器就会恢复到不能访问的状态。内核版本是4.19, 10月16日晚编译, 事前已重新configure。
> 
> 看来这个bug还有随机性的，这就很骚了

不是随机的，这个问题是很轻易复现的

coolsnowwolf, 2019-10-16:
我一次都没有重现过，或者你们用我编译的 IMG 测试即可

coolsnowwolf, 2019-10-16:
PS: 不用轻易换内核，最近的 patch 都只针对 Kernel 4.9 in x64 ,如果你们换了就要自己解决了

CirnoSalt, 2019-10-16:
> 并不存在随机的问题，请更新你们的代码， 这个在线音乐播放器 in Docker 可以完整运行就是实例。
> 
> 请 rm -rf build_dir ， rm .config, rm -rf tmp 后，更新 feeds 后编译

好的，我明天尝试更新至今晚的编译版本，调换WAN/LAN的物理接口再做测试。

KFERMercer, 2019-10-16:
> > 并不存在随机的问题，请更新你们的代码， 这个在线音乐播放器 in Docker 可以完整运行就是实例。
> > 请 rm -rf build_dir ， rm .config, rm -rf tmp 后，更新 feeds 后编译
> 
> 好的，我明天尝试更新至今晚的编译版本，调换WAN/LAN的物理接口再做测试。

你重启一下防火墙再重新用centos进行ping测试一下试试能不能正常联网，顺便问一下你的内核版本是？

CirnoSalt, 2019-10-16:
> > > 并不存在随机的问题，请更新你们的代码， 这个在线音乐播放器 in Docker 可以完整运行就是实例。
> > > 请 rm -rf build_dir ， rm .config, rm -rf tmp 后，更新 feeds 后编译
> > 
> > 
> > 好的，我明天尝试更新至今晚的编译版本，调换WAN/LAN的物理接口再做测试。
> 
> 试试重启防火墙再重新在centos下试试能不能正常联网，顺便问一下你的内核版本是？

4.9.196

CirnoSalt, 2019-10-17:
> 并不存在随机的问题，请更新你们的代码， 这个在线音乐播放器 in Docker 可以完整运行就是实例。
> 
> 请 rm -rf build_dir ， rm .config, rm -rf tmp 后，更新 feeds 后编译

已经使用昨晚的版本(4.19.79)测试，并且调换了LAN/WAN口，没有做任何其他设置，容器内仍然无法访问外网，但ping外网时返回了Destination Port Unreachable错误，之前是超时。
[image]
（已尝试重启防火墙，无效）

CirnoSalt, 2019-10-17:
更新：
1：vim /etc/sysctl.conf，添加下列内容：
```
net.bridge.bridge-nf-call-ip6tables = 1
net.bridge.bridge-nf-call-iptables = 1
net.bridge.bridge-nf-call-arptables = 1
```
2：Luci>网络>防火墙>转发：接受
3：Luci>状态>防火墙>重启防火墙
4：ssh执行service dockerd restart
故障排除。

zhg-em, 2019-10-17:
完全没有问题，很正常！
uubuntu的：
[image]

centos的：
[image]

closed 2019-10-18 by coolsnowwolf

Yrp, 2020-07-21:
> 更新：
> 1：vim /etc/sysctl.conf，添加下列内容：
> 
> ```
> net.bridge.bridge-nf-call-ip6tables = 1
> net.bridge.bridge-nf-call-iptables = 1
> net.bridge.bridge-nf-call-arptables = 1
> ```
> 
> 2：Luci > 网络 > 防火墙 > 转发：接受
> 3：Luci > 状态 > 防火墙 > 重启防火墙
> 4：ssh 执行 service dockerd restart
> 故障排除。

非常感谢，遇到和你一样的问题，按照你的操作解决了

rambodoctor, 2021-06-23:
> 更新：
> 1：vim /etc/sysctl.conf，添加下列内容：
> 
> ```
> net.bridge.bridge-nf-call-ip6tables = 1
> net.bridge.bridge-nf-call-iptables = 1
> net.bridge.bridge-nf-call-arptables = 1
> ```
> 
> 2：Luci>网络>防火墙>转发：接受
> 3：Luci>状态>防火墙>重启防火墙
> 4：ssh执行service dockerd restart
> 故障排除。

21年6 月  出现这个问题 重装个n次  花费4天时间   最终照着这个解决  特地登录感谢

uclee888, 2022-07-23:
> 更新： 1：vim /etc/sysctl.conf，添加下列内容：
> 
> ```
> net.bridge.bridge-nf-call-ip6tables = 1
> net.bridge.bridge-nf-call-iptables = 1
> net.bridge.bridge-nf-call-arptables = 1
> ```
> 
> 2：Luci>网络>防火墙>转发：接受 3：Luci>状态>防火墙>重启防火墙 4：ssh执行service dockerd restart 故障排除。

改成net.bridge.bridge-nf-call-ip6tables = 1
net.bridge.bridge-nf-call-iptables = 1
net.bridge.bridge-nf-call-arptables = 1
之后会引入新的问题，就是nat回环失效。简单说就是设备在内网无法通过路由器公网IP+端口（比如DDNS）访问内网映射出去的另一个设备，比如监控摄像机

whazd7, 2023-01-12:
> ，并且调换了LAN/WAN口，没有做任何其他设置，容器内仍然无法访问外网，但ping外网

非常非常感谢，解决了困扰我许久的问题

upsangel, 2023-09-01:
> 20220723 更新 实测，net.bridge.bridge-nf-call-arptables = 0 net.bridge.bridge-nf-call-ip6tables = 0 net.bridge.bridge-nf-call-iptables = 0 的情况下，只要开启Luci>网络>防火墙>转发：接受 即可使bridge模式的容器联网

我也是打開 Luci>网络>防火墙>转发：接受 即可使bridge模式的容器联网
但問題是，這個設定原本是 REJECT的，全盤接受也就是將所有跨zone的forward都允許，安全風險不會變得超級大？

jieran233, 2023-09-07:
> 更新： 1：vim /etc/sysctl.conf，添加下列内容：
> 
> ```
> net.bridge.bridge-nf-call-ip6tables = 1
> net.bridge.bridge-nf-call-iptables = 1
> net.bridge.bridge-nf-call-arptables = 1
> ```
> 
> 2：Luci>网络>防火墙>转发：接受 3：Luci>状态>防火墙>重启防火墙 4：ssh执行service dockerd restart 故障排除。

用了您的方法一次成功，非常感谢！！

quchifanma, 2023-09-07:
> 更新： 1：vim /etc/sysctl.conf，添加下列内容：
> 
> ```
> net.bridge.bridge-nf-call-ip6tables = 1
> net.bridge.bridge-nf-call-iptables = 1
> net.bridge.bridge-nf-call-arptables = 1
> ```
> 
> 2：Luci>网络>防火墙>转发：接受 3：Luci>状态>防火墙>重启防火墙 4：ssh执行service dockerd restart 故障排除。

确实好用，这的非常感谢，这个也折磨了我很多天。