coolsnowwolf/lede #1763: 【luci-app-ssr-plus】V2RAY订阅bug

wekingchen, 2019-10-16:
当前版本在读取V2RAY订阅URL地址时，如果当Transport类型为HTTP/2的时候，HTTP/2 Host和HTTP/2 Path的值无法写入，表现为留空。
不知道是否能修正一下？谢谢！

wekingchen, 2019-10-17:
比如： http://v2.limengtongjian.pw/v2.txt

是从v2rayN导出的链接：
vmess://ew0KICAidiI6ICIyIiwNCiAgInBzIjogIkhLX1FFWFciLA0KICAiYWRkIjogInd3dy5hYWEuY29tIiwNCiAgInBvcnQiOiAiNDQzIiwNCiAgImlkIjogIjYwYjdiNDBmLTQ5YTItNDEyOS04ZDUzLTk4NTQ2OTUwMTlmMCIsDQogICJhaWQiOiAiNjQiLA0KICAibmV0IjogImgyIiwNCiAgInR5cGUiOiAibm9uZSIsDQogICJob3N0IjogImFhYS5jb20iLA0KICAicGF0aCI6ICIvYWRtaW4iLA0KICAidGxzIjogInRscyINCn0=

BASE64加密后的内容：
dm1lc3M6Ly9ldzBLSUNBaWRpSTZJQ0l5SWl3TkNpQWdJbkJ6SWpvZ0lraExYMUZGV0ZjaUxBMEtJQ0FpWVdSa0lqb2dJbmQzZHk1aFlXRXVZMjl0SWl3TkNpQWdJbkJ2Y25RaU9pQWlORFF6SWl3TkNpQWdJbWxrSWpvZ0lqWXdZamRpTkRCbUxUUTVZVEl0TkRFeU9TMDRaRFV6TFRrNE5UUTJPVFV3TVRsbU1DSXNEUW9nSUNKaGFXUWlPaUFpTmpRaUxBMEtJQ0FpYm1WMElqb2dJbWd5SWl3TkNpQWdJblI1Y0dVaU9pQWlibTl1WlNJc0RRb2dJQ0pvYjNOMElqb2dJbUZoWVM1amIyMGlMQTBLSUNBaWNHRjBhQ0k2SUNJdllXUnRhVzRpTEEwS0lDQWlkR3h6SWpvZ0luUnNjeUlOQ24wPQ==

导入后HTTP/2 Host和HTTP/2 Path的值不能正确填入。

dongfangy2011, 2019-10-19:
经常出现如下现象：
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute function dispatcher target for entry '/admin/services/shadowsocksr'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute cbi dispatcher target for entry '/admin/services/shadowsocksr/client'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/shadowsocksr/client.lua:18: bad argument #1 to 'upper' (string expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>

coolsnowwolf, 2019-10-19:
bad argument #1 to 'upper'  是订阅的时候，输入了错误的类型，估计是 ss 之类了

wekingchen, 2019-10-19:
@dongfangy2011 大兄dei我报bug麻烦别歪楼好不，你可以自己开一贴啊。这样影响老大定位原有问题了！

coolsnowwolf, 2019-10-19:
已经解决了，你用新版试试

closed 2019-10-19 by coolsnowwolf