coolsnowwolf/lede #1764: 编译出错

xinxijishuwyq, 2019-10-17:
Package docker-ce wants to install file /home/ken/lede/build_dir/target-x86_64_musl/root-x86/etc/init.d/dockerd
        But that file is already provided by package  * luci-app-docker
 * opkg_install_cmd: Cannot install package docker-ce

coolsnowwolf, 2019-10-17:
make clean

closed 2019-10-18 by coolsnowwolf

mingfanwang, 2020-01-12:
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-docker:
 *      kmod-nf-ipvs * 
 * opkg_install_cmd: Cannot install package luci-app-docker.
 * check_data_file_clashes: Package docker-ce wants to install file /home/wmf181830544/friendlywrt-h5/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/root-sunxi/etc/init.d/dockerd
        But that file is already provided by package  * luci-app-docker
 * opkg_install_cmd: Cannot install package docker-ce.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/wmf181830544/friendlywrt-h5/friendlywrt'
package/Makefile:104: recipe for target '/home/wmf181830544/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_install' failed
make[1]: *** [/home/wmf181830544/friendlywrt-h5/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/wmf181830544/friendlywrt-h5/friendlywrt'
/home/wmf181830544/friendlywrt-h5/friendlywrt/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2