coolsnowwolf/lede #1769: qbittorrent无法启动

hong0980, 2019-10-17:
依赖应该都安装了，下载路径也正确。/etc/init.d/qbittorrent文件权限也正确。qbittorrent-nox命令启动WebUI无法打开。L大求解，谢谢！
root@OpenWrt:~# ldd /usr/bin/qbittorrent-nox
        /lib/ld-musl-mipsel-sf.so.1 (0x77ee2000)
        libz.so.1 => /usr/lib/libz.so.1 (0x77ec2000)
        libtorrent-rasterbar.so.9 => /usr/lib/libtorrent-rasterbar.so.9 (0x77aee000)
        libboost_system.so.1.69.0 => /usr/lib/libboost_system.so.1.69.0 (0x77adc000)
        libQt5Network.so.5 => /usr/lib/libQt5Network.so.5 (0x77912000)
        libQt5Xml.so.5 => /usr/lib/libQt5Xml.so.5 (0x778c4000)
        libQt5Core.so.5 => /usr/lib/libQt5Core.so.5 (0x77256000)
        libatomic.so.1 => /lib/libatomic.so.1 (0x77241000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x770e9000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x770c5000)
        libc.so => /lib/ld-musl-mipsel-sf.so.1 (0x77ee2000)
        libssl.so.1.0.0 => /usr/lib/libssl.so.1.0.0 (0x7705c000)
        libcrypto.so.1.0.0 => /usr/lib/libcrypto.so.1.0.0 (0x76eb5000)
        libboost_chrono.so.1.69.0 => /usr/lib/libboost_chrono.so.1.69.0 (0x76e9b000)
        libboost_random.so.1.69.0 => /usr/lib/libboost_random.so.1.69.0 (0x76e82000)
        libicui18n.so.61 => /usr/lib/libicui18n.so.61 (0x76c36000)
        libicuuc.so.61 => /usr/lib/libicuuc.so.61 (0x76ac6000)
        libicudata.so.61 => /usr/lib/libicudata.so.61 (0x76ab4000)

coolsnowwolf, 2019-10-17:
qbt 新版不再支持 mips 了，需要更换arm 或者其他

hong0980, 2019-10-17:
> qbt 新版不再支持 mips 了，需要更换arm 或者其他

谢谢回复！L大4.1.5和4.1.3也不能启动，不知是什么原因。是启动脚本有问题吗？

coolsnowwolf, 2019-10-18:
你用 x64 版本测试下

zhg-em, 2019-10-19:
x64 可以用的。但打开web管理界面按钮一直有问题！

hong0980, 2019-11-01:
L大mips架构的D2在qBittorrent-4.1.9 +qt5-5.8 +rblibtorrent1.1.13 +boost-1.68 +libopenssl-1.1.1d 中能运行，估计是bootst版本的问题在MIPS中不能运行。

hong0980, 2019-11-01:
> web管理界面按钮

web管理界面按钮不灵是要点击两次应用保存后，再从浏览器直接输入网址和端口能进入。再从Qb里设置把“启用跨站点请求伪造(CSRF)保护”去掉保存。ok了。我是init.d下的文件加进去。点两次应用保存就能直接能从web管理界面按钮打开进入。

hong0980, 2019-11-01:
> x64 可以用的。但打开web管理界面按钮一直有问题！

web管理界面按钮不灵是要点击两次应用保存后，再从浏览器直接输入网址和端口能进入。再从Qb里设置把“启用跨站点请求伪造(CSRF)保护”去掉保存。ok了。我是init.d下的文件加进去。点两次应用保存就能直接能从web管理界面按钮打开进入。

closed 2019-11-28 by hong0980

fr0zenrain, 2019-12-31:
新版不支持mips是啥意思？4.2.1可以编过，但是下载文件好像老丢失损坏的。有官方通知么，能不能给个链接。