coolsnowwolf/lede #1775: 通过dns审计，httping和ping 自动加gfwlist ipset

rufengsuixing, 2019-10-18:
以下是 兼容smartdns的审计，dnsmasq需要修改
审计的log格式为
2019-10-18 08:20:42,982] 127.0.0.1 query kv701.prod.do.dsp.mp.microsoft.com, type 1, result 184.85.125.69
代码如下

#!/bin/sh
tail -f /tmp/log/smartdns-audit.log | awk  -F "[, ]+" '{
if ($8=="1" && a[$10] == "")
{
print("logger"$10"  "$6);
a[$10]=$6;
"ipset test gfwlist "$10" 2>&1"| getline ipset;
print(ipset)
ret = system("ping -A -c 2 -W 5 "$10" >/dev/null");
if (ret!=0){
if (index(ipset,"Warning")==0){
ret = system("httping -c 1 -t 5 -q "$6);
print("time"ret)
if (ret!=0){
print("logger autoaddip "$10);
system("ipset add gfwlist "$10);
}
}
}
}
}'

超时可以调整加快探测速度
如果有http代理，可以完善逻辑，防止误加列表
异步加入，希望哪个大佬有时间能够用更好的实现完成

coolsnowwolf, 2019-10-18:
smartdns 下，App Store，一些P2SP下载什么的只能获取到一个ip，不能完成多 CDN 多并发下载，导致下载速度反而更慢

rufengsuixing, 2019-10-19:
抛开smartdns，使用审计+主动探测这样的方式加入ipset列表我觉得是可行的，新使用了nf_conntrack减少ipset列表ip来回切换的可能

#!/bin/sh
tail -F /tmp/log/smartdns-audit.log | awk  -F "[, ]+" '{
if ($8=="1" && (!($10 in a)))
{
print("logger "$10"  "$6); 
"ipset test gfwlist "$10" 2>&1"| getline ipset;
close("ipset test gfwlist "$10" 2>&1");
if (index(ipset,"Warning")!=0){
print("pass");
next;
}

tryhttps=0;
tryhttp=0;
while ("grep "$10" /proc/net/nf_conntrack"|getline ret > 0)
{
    split(ret, b);
    if (b[8]=="dst="$10)
    {
        if (b[10]=="dport=443"){
            tryhttps=1;
            break;
        }
        else if (b[10]=="dport=80"){
            tryhttp=1;
            break;
        }
    }
}
close("grep "$10" /proc/net/nf_conntrack");
if (tryhttps==1)
{
cmd=("httping -c 1 -t 3 -l "$6);
print("443");
}
else if (tryhttp==1)
{
cmd=("httping -c 1 -t 3 "$6);
print("80");
}else
{
print("no 443 80 found");
next;
}
a[$10]=$6;
addlist=0;
while ((cmd | getline ret) > 0)
{
    if (addlist==0){
    if (index(ret,"short read")!=0)
    {
    system("ipset add gfwlist "$10);
    print("doname rst autoaddip "$10);
    addlist=1;
    }
    else if (index(ret,"timeout")!=0)
    {
        print("direct so slow"$10" "$6);
        close(cmd);
        next;
    }
    }
}
close(cmd);
split(ret, c,"[ /]+");
print(c[6]);
if (addlist==0 && c[6]=="failed,")
{
    system("ipset add gfwlist "$10);
    print("can not connect autoaddip "$10);
    addlist=1;
}
if (addlist==1){
while ((cmd | getline ret) > 0)
{
    if (index(ret,"short read")!=0)
    {
    system("ipset del gfwlist "$10);
    print("doname proxy rst autodelip "$10);
    close(cmd);
    next;
    }
    else if (index(ret,"timeout")!=0)
    {
        print("proxy so slow"$10" "$6);
        close(cmd);
        next;
    }
}
close(cmd);
split(ret, c,"[ /]+");
print(c[6]);
if (c[6]=="failed,")
{
    system("ipset del gfwlist "$10);
    print("proxy can not connect autodelip "$10);
}}}}'

rufengsuixing, 2019-10-20:
[](https://github.com/rufengsuixing/auto-ipsetadder-for-openwrt)

closed 2019-10-20 by rufengsuixing