coolsnowwolf/lede #1791: 生成固件报错信息，请教

Aes64X, 2019-10-19:
错误信息，-j1下

ubinize: volume size was not specified in section "kernel", assume minimum to fit image "/home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/p2w_r619ac-fit-uImage.itb"2574552 bytes (2.4 MiB)
ubinize: volume size was not specified in section "rootfs", assume minimum to fit image "/home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/root.squashfs"14564452 bytes (13.8 MiB)
cat /home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-p2w_r619ac-squashfs-nand-factory.ubi.tmp >> /home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-p2w_r619ac-squashfs-nand-factory.ubi
rm /home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-p2w_r619ac-squashfs-nand-factory.ubi.tmp
cp /home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-p2w_r619ac-squashfs-nand-factory.ubi /home/vbeta/lede/bin/targets/ipq40xx/generic/openwrt-ipq40xx-generic-p2w_r619ac-squashfs-nand-factory.ubi
[ -f /home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/p2w_r619ac-fit-uImage.itb -a -f /home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/root.squashfs ]
sh /home/vbeta/lede/scripts/sysupgrade-tar.sh --board p2w_r619ac --kernel /home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/p2w_r619ac-fit-uImage.itb --rootfs /home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/root.squashfs /home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-p2w_r619ac-squashfs-nand-sysupgrade.bin
tar: Option --mtime: Treating date '@1571425423' as 2019-10-18 19:03:43
sysupgrade-p2w_r619ac/
sysupgrade-p2w_r619ac/CONTROL
sysupgrade-p2w_r619ac/kernel
sysupgrade-p2w_r619ac/root
echo '{  "supported_devices":["p2w,r619ac"], "version": { "dist": "OpenWrt", "version": "SNAPSHOT", "revision": "r1548-1328cae", "board": "ipq40xx" } }' | fwtool -I - /home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-p2w_r619ac-squashfs-nand-sysupgrade.bin
[ ! -s "/home/vbeta/lede/key-build" -o ! -s "/home/vbeta/lede/key-build.ucert" -o ! -s "/home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-p2w_r619ac-squashfs-nand-sysupgrade.bin" ] || { cp "/home/vbeta/lede/key-build.ucert" "/home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-p2w_r619ac-squashfs-nand-sysupgrade.bin.ucert" ; usign -S -m "/home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-p2w_r619ac-squashfs-nand-sysupgrade.bin" -s "/home/vbeta/lede/key-build" -x "/home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-p2w_r619ac-squashfs-nand-sysupgrade.bin.sig" ; ucert -A -c "/home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-p2w_r619ac-squashfs-nand-sysupgrade.bin.ucert" -x "/home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-p2w_r619ac-squashfs-nand-sysupgrade.bin.sig" ; fwtool -S "/home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-p2w_r619ac-squashfs-nand-sysupgrade.bin.ucert" "/home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-p2w_r619ac-squashfs-nand-sysupgrade.bin" ; }
cp /home/vbeta/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-p2w_r619ac-squashfs-nand-sysupgrade.bin /home/vbeta/lede/bin/targets/ipq40xx/generic/openwrt-ipq40xx-generic-p2w_r619ac-squashfs-nand-sysupgrade.bin
Makefile:444: *** Missing Build/qsdk-ipq-factory-nand.  Stop.
make[5]: Leaving directory '/home/vbeta/lede/target/linux/ipq40xx/image'
Makefile:24: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/vbeta/lede/target/linux/ipq40xx'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/vbeta/lede/target/linux'
time: target/linux/install#43.97#6.46#33.36
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/home/vbeta/lede'
target/Makefile:19: recipe for target '/home/vbeta/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install' failed
make[1]: *** [/home/vbeta/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/vbeta/lede'
/home/vbeta/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

bin文件夹中已出现*.bin和*.ubi

coolsnowwolf, 2019-10-19:
请找源代码作者

closed 2019-10-19 by coolsnowwolf

Aes64X, 2019-10-19:
> 
> 
> 请找源代码作者

我编译成功了，感谢您和LGA1150的源码。我看到了JS对这玩意进行了涨价，我同样不会公布这一部分，直到这个设备被忘记。