coolsnowwolf/lede #1796: linksys wrt32x从原厂固件刷入openwrt，以后无法升级固件

ITdesk01, 2019-10-19:
情况如下
 网购了一台Linksys WRT32X，通过原厂固件将其刷成双分区，可以切换原厂和openwrt，后面刷了R9固件无法升级

报错如下
`root@OpenWrt:~# /usr/sbin/fw_printenv -n boot_part
Cannot parse config file '/etc/fw_env.config': No such file or directory`
`root@OpenWrt:/# Watchdog handover: fd=3
- watchdog -
killall: telnetd: no process killed
killall: dropbear: no process killed
Sending TERM to remaining processes ... ubusd logd rpcd netifd crond vsftpd dbus-daemon hostapd hostapd hostapd nlbwmon avahi-daemon vlmcsd dnsmasq sh sleep smbd ntpd smbd-notifyd cleanupd ttyd wsdd2
[  296.880606] device wlan1 left promiscuous mode
[  296.885112] br-lan: port 2(wlan1) entered disabled state
[  297.015252] device wlan0 left promiscuous mode
[  297.019750] br-lan: port 3(wlan0) entered disabled state
[  297.050281] device wlan2 left promiscuous mode
[  297.054766] br-lan: port 4(wlan2) entered disabled state
Sending KILL to remaining processes ...
Switching to ramdisk...
[  301.526412] UBIFS (ubi0:1): background thread "ubifs_bgt0_1" stops
[  301.598029] UBIFS (ubi0:1): un-mount UBI device 0
Performing system upgrade...
Cannot parse config file '/etc/fw_env.config': No such file or d[  301.625353] reboot: Restarting system`

报错引起原因
`/etc/fw_env.config`

fw_env.config内容
`/dev/mtd1 0x0 0x20000 0x20000`

临时解决办法
新建/etc/fw_env.config，然后内容填/dev/mtd1 0x0 0x20000 0x20000，这样就可以切换回原厂分区，然后可以升级，缺点luci-app-advanced-reboot无法正常工作

coolsnowwolf, 2019-10-19:
已修复

ITdesk01, 2019-10-19:
20191019新版的固件
1.解决了/etc/fw_env.config问题，但原版固件分区变成了2， R9固件分区变成1
2.openwrt官方源码的原版分区应该是1，openwrt是2才对
3.luci-app-advanced-reboot依旧无法使用
4.可以刷入固件升级了，但缺点暴露出来了，我在R9刷入openwrt固件，会导致在分区2的原版固件直接被覆盖，变成双openwrt，回到R9分区1，重新刷入R9固件，查看分区直接变成2，也就是说此版本的固件分区位置不对，刷入此版本后会导致双openwrt固件 

导致刷入分区不是2，而是1，可能是以下代码
`killall: watchdog: no process killed
Sending TERM to remaining processes ... logd rpcd netifd odhcpd crond siproxd uhttpd dnsmasq /sbin/sysupgrade: line 237: can't open /proc/5619/stat: no such file
/sbin/sysupgrade: line 237: can't open /proc/5616/cmdline: no such file
dnsmasq ntpd ubusd
Sending KILL to remaining processes ...
Switching to ramdisk...
Performing system upgrade...
Unlocking kernel1 ...
Erasing kernel1 ...
Unlocking kernel1 ...

Writing from <stdin> to kernel1 ...
Upgrade completed
Rebooting system...`

coolsnowwolf, 2019-10-21:
现在这个逻辑是正确的
分区1 的系统输入分区 2 ，原厂就是这样的设定，这样起码可以保证你能进入目前的系统中

luci 问题是没有更新支持 wrt32x，你提交一个pr即可

closed 2019-10-21 by coolsnowwolf

ITdesk01, 2019-10-21:
问题是原厂刷入R9，R9的分区不应该是2吗，为什么R9的分区变成1，而原厂固件分区变成2

我尝试在R9分区1的时候升级固件，直接导致分区2的原厂固件直接覆盖，这样就不是一原厂，一第三方了

我也测试过openwrt官方固件，默认从原厂刷入openwrt，重启完成以后openwrt所在分区是2，而不是1