coolsnowwolf/lede #1801: raspberry pi 3b + 编译时一些看不懂的提示

blackjelly, 2019-10-19:
```
Cortex-A53: 826319: System might deadlock if a write cannot complete until read data is accepted (ARM64_ERRATUM_826319) [Y/n/?] y
Cortex-A53: 827319: Data cache clean instructions might cause overlapping transactions to the interconnect (ARM64_ERRATUM_827319) [Y/n/?] y
Cortex-A53: 824069: Cache line might not be marked as clean after a CleanShared snoop (ARM64_ERRATUM_824069) [Y/n/?] y
Cortex-A53: 819472: Store exclusive instructions might cause data corruption (ARM64_ERRATUM_819472) [Y/n/?] y
Cortex-A57: 832075: possible deadlock on mixing exclusive memory accesses with device loads (ARM64_ERRATUM_832075) [Y/n/?] y
Cortex-A53: 843419: A load or store might access an incorrect address (ARM64_ERRATUM_843419) [Y/n/?] y
Cortex-A55: 1024718: Update of DBM/AP bits without break before make might result in incorrect update (ARM64_ERRATUM_1024718) [N/y/?] n
Cortex-A76: Software Step might prevent interrupt recognition (ARM64_ERRATUM_1463225) [Y/n/?] (NEW) 

choice[1-4?]: 1
Enable seccomp to safely compute untrusted bytecode (SECCOMP) [N/y/?] n
Enable paravirtualization code (PARAVIRT) [N/y/?] n
Paravirtual steal time accounting (PARAVIRT_TIME_ACCOUNTING) [N/y/?] n
kexec system call (KEXEC) [N/y/?] n
Build kdump crash kernel (CRASH_DUMP) [N/y/?] n
Xen guest support on ARM64 (XEN) [N/y/?] n
Unmap kernel when running in userspace (aka "KAISER") (UNMAP_KERNEL_AT_EL0) [Y/n/?] y
Harden the branch predictor against aliasing attacks (HARDEN_BRANCH_PREDICTOR) [Y/n/?] y
Harden EL2 vector mapping against system register leak (HARDEN_EL2_VECTORS) [Y/n/?] y
Speculative Store Bypass Disable (ARM64_SSBD) [Y/n/?] y
Emulate Privileged Access Never using TTBR0_EL1 switching (ARM64_SW_TTBR0_PAN) [N/y/?] (NEW) 

choice[1-4?]: 1
Enable seccomp to safely compute untrusted bytecode (SECCOMP) [N/y/?] n
Enable paravirtualization code (PARAVIRT) [N/y/?] n
Paravirtual steal time accounting (PARAVIRT_TIME_ACCOUNTING) [N/y/?] n
kexec system call (KEXEC) [N/y/?] n
Build kdump crash kernel (CRASH_DUMP) [N/y/?] n
Xen guest support on ARM64 (XEN) [N/y/?] n
Unmap kernel when running in userspace (aka "KAISER") (UNMAP_KERNEL_AT_EL0) [Y/n/?] y
Harden the branch predictor against aliasing attacks (HARDEN_BRANCH_PREDICTOR) [Y/n/?] y
Harden EL2 vector mapping against system register leak (HARDEN_EL2_VECTORS) [Y/n/?] y
Speculative Store Bypass Disable (ARM64_SSBD) [Y/n/?] y
Emulate Privileged Access Never using TTBR0_EL1 switching (ARM64_SW_TTBR0_PAN) [N/y/?] (NEW) 
ARM Scalable Vector Extension support (ARM64_SVE) [Y/n/?] y
Randomize the address of the kernel image (RANDOMIZE_BASE) [N/y/?] n

*
* Performance monitor support
*
ARM CCI PMU driver (ARM_CCI_PMU) [N/m/y/?] n
ARM CCN driver support (ARM_CCN) [N/m/y/?] n
ARM PMU framework (ARM_PMU) [Y/n/?] y
ARM DynamIQ Shared Unit (DSU) PMU (ARM_DSU_PMU) [N/m/y/?] n
Enable support for the ARMv8.2 Statistical Profiling Extension (ARM_SPE_PMU) [N/m/y/?] n
RaspberryPi AXI Performance monitors (RPI_AXIPERF) [N/m/y/?] (NEW) 

A4 tech mice (HID_A4TECH) [N/m/?] n
Accutouch touch device (HID_ACCUTOUCH) [N/m/?] n
ACRUX game controller support (HID_ACRUX) [N/m/?] n
Apple {i,Power,Mac}Books (HID_APPLE) [N/m/?] n
Apple infrared receiver (HID_APPLEIR) [N/m/?] n
Asus (HID_ASUS) [N/m/?] n
Aureal (HID_AUREAL) [N/m/?] n
Belkin Flip KVM and Wireless keyboard (HID_BELKIN) [N/m/?] n
Betop Production Inc. force feedback support (HID_BETOP_FF) [N/m/?] n
BigBen Interactive Kids' gamepad support (HID_BIGBEN_FF) [N/m/?] (NEW) 
```

outemancut, 2019-10-19:
自己解决不了就别乱改内核版本！

blackjelly, 2019-10-19:
> 自己解决不了就别乱改内核版本！

自己并没有做修改

ligl0702, 2019-10-20:
我这边编译也出现了这些提示，目前我都是一路回车的。可以编译通过的。

closed 2019-10-20 by coolsnowwolf