coolsnowwolf/lede #1808: SSR Plus v2ray订阅节点无法解析

shang-he, 2019-10-20:
在pc上用v2rayN可以订阅成功 在ssr Plus 上更新无法显示节点(pc显示30多个节点) 下边是日志
`2019-10-20 08:11:38 V2Ray 4.20.0 (V2Fly, a community-driven edition of V2Ray.) Lean Custom Build Started!
2019-10-20 08:09:40 V2Ray 4.20.0 (V2Fly, a community-driven edition of V2Ray.) Lean Custom Build Started!
shadowsocksr_subscribe[16047]: default 服务器订阅更新成功 服务器数量: 1 新增服务器: 1 删除服务器: 1
2019-10-20 08:09:38:在线订阅列表更新完成!请等待网页自动刷新!
2019-10-20 08:09:38:本次更新订阅来源 【default】 服务器数量: 1 新增服务器: 1 删除服务器: 1
uci: Invalid argument
	-f FILE	Read pattern from file
	-e PTRN	Pattern to match
	-C N	Same as '-A N -B N'
	-B N	Print N lines of leading context
	-A N	Print N lines of trailing context
	-m N	Match up to N times per file
	-E	PATTERN is an extended regexp
	-F	PATTERN is a literal (not regexp)
	-x	Match whole lines only
	-w	Match whole words only
	-i	Ignore case
	-r	Recurse
	-s	Suppress open and read errors
	-v	Select non-matching lines
	-q	Quiet. Return 0 if PATTERN is found, 1 otherwise
	-o	Show only the matching part of line
	-c	Show only count of matching lines
	-L	Show only names of files that don't match
	-l	Show only names of files that match
	-n	Add 'line_no:' prefix
	-h	Do not add 'filename:' prefix
	-H	Add 'filename:' prefix

Search for PATTERN in FILEs (or stdin)

Usage: grep [-HhnlLoqvsriwFE] [-m N] [-A/B/C N] PATTERN/-e PATTERN.../-f FILE [FILE]...

BusyBox v1.30.0 () multi-call binary.
	-f FILE	Read pattern from file
	-e PTRN	Pattern to match
	-C N	Same as '-A N -B N'
	-B N	Print N lines of leading context
	-A N	Print N lines of trailing context
	-m N	Match up to N times per file
	-E	PATTERN is an extended regexp
	-F	PATTERN is a literal (not regexp)
	-x	Match whole lines only
	-w	Match whole words only
	-i	Ignore case
	-r	Recurse
	-s	Suppress open and read errors
	-v	Select non-matching lines
	-q	Quiet. Return 0 if PATTERN is found, 1 otherwise
	-o	Show only the matching part of line
	-c	Show only count of matching lines
	-L	Show only names of files that don't match
	-l	Show only names of files that match
	-n	Add 'line_no:' prefix
	-h	Do not add 'filename:' prefix
	-H	Add 'filename:' prefix

Search for PATTERN in FILEs (or stdin)

Usage: grep [-HhnlLoqvsriwFE] [-m N] [-A/B/C N] PATTERN/-e PATTERN.../-f FILE [FILE]...

BusyBox v1.30.0 () multi-call binary.
2019-10-20 08:09:38:节点：【】
uci: Entry not found
base64: invalid input
2019-10-20 08:09:38:共计1个节点
base64: invalid input

2019-10-20 08:09:38 (546 KB/s) - written to stdout [26]

     0K                                                         546K=0s

Saving to: 'STDOUT'
Length: unspecified [text/html]
HTTP request sent, awaiting response... 200 OK
Connecting to  connected.
Resolving  ...
--2019-10-20 08:09:30--  
2019-10-20 08:09:29:开始下载订阅链接到本地临时文件，请稍等...
2019-10-20 08:09:29:开始更新在线订阅列表...
2019-10-20 08:09:29:从  获取订阅
2019-10-20 08:09:01 V2Ray 4.20.0 (V2Fly, a community-driven edition of V2Ray.) Lean Custom Build Started!
`

Kinodi, 2019-10-20:
[image]
z正常订阅

shang-he, 2019-10-20:
之前的机场订阅是可以用的 现在新买了一个机场不可以用了

shang-he, 2019-10-20:
你是这个模块的开发者吗 需要的话我可以吧订阅节点发给你

stl88083365, 2019-10-20:
订阅是可以的，你那个应该是机场的问题

stl88083365, 2019-10-20:
还有你确定是L大的源码无更改编译的，而不是那些魔改过的~！

shang-he, 2019-10-20:
固件是我自己clone的代码编译的 我问一下机场那边

shang-he, 2019-10-20:
重置订阅链接之后好了

closed 2019-10-20 by shang-he

wuyz320, 2022-08-06:
我编译的也无法进行订阅，好像是解析出了问题，手工添加节点是可以的，也能用，就是解析不出订阅网站下载的内容