coolsnowwolf/lede #1826: 树莓派 4 WIFI 依然无法工作

SuLingGG, 2019-10-21:
如题。拉取最新源码：

858d11ebc2df4ace6787e6215618f71f2de83f35

使用 2711 默认配置编译后树莓派 4 WIFI 依然无法工作...
另外 2709 的配置编译出来的固件的 WIFI 好像也有问题。
下面是使用 2711 默认配置编译的固件中的系统日志：
```
Mon Oct 21 17:50:59 2019 daemon.notice netifd: Interface 'lan' is enabled
Mon Oct 21 17:50:59 2019 daemon.notice netifd: Interface 'lan' is setting up now
Mon Oct 21 17:50:59 2019 daemon.notice netifd: Interface 'lan' is now up
Mon Oct 21 17:50:59 2019 daemon.notice netifd: Interface 'loopback' is enabled
Mon Oct 21 17:50:59 2019 daemon.notice netifd: Interface 'loopback' is setting up now
Mon Oct 21 17:50:59 2019 daemon.notice netifd: Interface 'loopback' is now up
Mon Oct 21 17:50:59 2019 daemon.notice netifd: Network device 'lo' link is up
Mon Oct 21 17:50:59 2019 daemon.notice netifd: Interface 'loopback' has link connectivity
Mon Oct 21 17:50:59 2019 daemon.notice netifd: radio0 (1215): command failed: No error information (-524)
Mon Oct 21 17:50:59 2019 user.notice mac80211: Failed command: iw dev wlan0 del
Mon Oct 21 17:50:59 2019 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Mon Oct 21 17:50:59 2019 user.notice mac80211: Failed command: iw phy phy0 set antenna 0xffffffff 0xffffffff
Mon Oct 21 17:50:59 2019 daemon.notice netifd: radio0 (1215): command failed: I/O error (-5)
Mon Oct 21 17:50:59 2019 user.notice mac80211: Failed command: iw phy phy0 set distance 0
Mon Oct 21 17:50:59 2019 daemon.notice netifd: radio0 (1215): command failed: Too many open files in system (-23)
Mon Oct 21 17:50:59 2019 user.notice mac80211: Failed command: iw phy phy0 interface add wlan0 type __ap
Mon Oct 21 17:50:59 2019 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Mon Oct 21 17:50:59 2019 kern.info kernel: [   10.097074] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Mon Oct 21 17:50:59 2019 kern.info kernel: [   10.108568] br-lan: port 2(wlan0) entered blocking state
Mon Oct 21 17:50:59 2019 kern.info kernel: [   10.113971] br-lan: port 2(wlan0) entered disabled state
Mon Oct 21 17:50:59 2019 kern.info kernel: [   10.119547] device wlan0 entered promiscuous mode
Mon Oct 21 17:50:59 2019 kern.info kernel: [   10.124388] br-lan: port 2(wlan0) entered blocking state
Mon Oct 21 17:50:59 2019 kern.info kernel: [   10.129703] br-lan: port 2(wlan0) entered forwarding state
Mon Oct 21 17:50:59 2019 kern.info kernel: [   10.135318] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
Mon Oct 21 17:50:59 2019 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon Oct 21 17:50:59 2019 daemon.notice netifd: bridge 'br-lan' link is up
Mon Oct 21 17:50:59 2019 daemon.notice netifd: Interface 'lan' has link connectivity
Mon Oct 21 17:50:59 2019 daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
Mon Oct 21 17:50:59 2019 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->ACS
Mon Oct 21 17:50:59 2019 daemon.notice hostapd: wlan0: ACS-STARTED
```

SuLingGG, 2019-10-21:
另外树莓派固件里 Docker 分区脚本中的 sda 应该改成 mmcblk ，望修复~
（这里就不单独提 Issue 了）

coolsnowwolf, 2019-10-21:
make clean

closed 2019-10-21 by coolsnowwolf