coolsnowwolf/lede #1827: dynv6

madcat0513, 2019-10-21:
已有ipv6和ipv4公网地址，用dynv6更新ipv4地址没问题，但是更新ipv6地址貌似脚本只会获取本地的ipv6地址。
日志如下：
 221353  info : Starting main loop at 2019-10-21 22:13
 221354       : Detect local IP on 'interface'
 221354       : Local IP 'fe80::2e30:33b8:5fa4:a836' detected on network 'wan'
 221354       : Update needed - L: 'fe80:0000:0000:0000:2e30:33b8:5fa4:a836' <> R: '240e:03a0:1600:0077:2e30:334d:49a4:a836'
 221355 ERROR : No or private or invalid IP 'fe80:0000:0000:0000:2e30:33b8:5fa4:a836' given! Please check your configuration
 221355 ERROR : No update send to DDNS Provider
 221355       : Waiting 600 seconds (Check Interval)
 222355       : Detect registered/public IP
 222356       : #> /usr/bin/nslookup madcat.dynv6.net  >/var/run/ddns/myddns_ipv6.dat 2>/var/run/ddns/myddns_ipv6.err
 222356       : Registered IP '240e:3a0:1600:77:2e30:334d:49a4:a836' detected
 222356  WARN : Updating IP at DDNS provider failed - starting retry 1/0

hanandjun, 2019-10-22:
可以尝试在高级设置里通过url或者接口获取ipv6地址

madcat0513, 2019-10-24:
现在只能通过url获取ipb6地址了，貌似获取接口地址的脚本有问题

closed 2019-12-30 by coolsnowwolf

hzx505, 2021-01-11:
刚使用几天的dynv6，也发现ipv6 ddns 错误日志