coolsnowwolf/lede #1835: 编译失败

f3404923, 2019-10-22:
make[3]: Leaving directory `/home/f3404923/lede/feeds/packages/utils/cgroupfs-mount'
time: package/feeds/packages/cgroupfs-mount/compile#0.06#0.02#0.40
make[3]: Entering directory `/home/f3404923/lede/feeds/packages/lang/golang/golang'
make[3]: Leaving directory `/home/f3404923/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#0.38#0.17#0.58
make[3]: Entering directory `/home/f3404923/lede/feeds/packages/utils/runc'
touch /home/f3404923/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.prepared_117f0ea1bc92ad180505bc663a5d76e9_6664517399ebbbc92a37c5bb081b5c53_check
. /home/f3404923/lede/include/shell.sh; gzip -dc /home/f3404923/lede/dl/runc-1.0.0-rc8+91-3e425f80.tar.gz | tar -C /home/f3404923/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.. -xf -
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
make[3]: *** [/home/f3404923/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.prepared_117f0ea1bc92ad180505bc663a5d76e9_6664517399ebbbc92a37c5bb081b5c53] Error 2
make[3]: Leaving directory `/home/f3404923/lede/feeds/packages/utils/runc'
time: package/feeds/packages/runc/compile#0.07#0.04#0.21
make[2]: *** [package/feeds/packages/runc/compile] Error 2
make[2]: Leaving directory `/home/f3404923/lede'
make[1]: *** [/home/f3404923/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/f3404923/lede'
make: *** [world] Error 2

-----------------------------------------------------------------
请问L大，这个怎么解啊？

ledoen, 2019-10-22:
感觉你前面还有一个错误啊，tar: Exiting with failure status due to previous errors

lphgor, 2019-10-22:
runc-1.0.0-rc8+91-3e425f80.tar.gz 这个文件没有下载成功吧？

ledoen, 2019-10-22:
下载失败的可能性比较大，我今天已经成功编译两次了，纯外网环境，失败过一次，提示我存储空间不够，硬盘调到20G就没问题了。

f3404923, 2019-10-22:
> runc-1.0.0-rc8+91-3e425f80.tar.gz 这个文件没有下载成功吧？

Thanks，我google了这个找不到该文件啊。make clean后重新编译还是同样错误

f3404923, 2019-10-23:
> 下载失败的可能性比较大，我今天已经成功编译两次了，纯外网环境，失败过一次，提示我存储空间不够，硬盘调到20G就没问题了。
确认过了全程外网连接没问题，磁盘空间也没问题，就是编译不过。重新编译前已经make clean了也不行

f3404923, 2019-10-23:
> > 下载失败的可能性比较大，我今天已经成功编译两次了，纯外网环境，失败过一次，提示我存储空间不够，硬盘调到20G就没问题了。

 确认过了全程外网连接没问题，磁盘空间也没问题，就是编译不过。重新编译前已经make clean了也不行

f3404923, 2019-10-23:
> 感觉你前面还有一个错误啊，tar: Exiting with failure status due to previous errors

查找了log，前面没有error 1出现，直接出的是error2. 看错误提示好像是加压文件失败，但是我找了一下也没有这个runc-1.0.0-rc8+91-3e425f80.tar.gz，google上也没有

coolsnowwolf, 2019-10-23:
runc 的 tar 下载失败，dl 里面删除相应的 tar 再来

yonggedebaqi, 2019-10-23:
我也是这个问题

yonggedebaqi, 2019-10-23:
大佬，能不能单独提供 runc 包文件。

yonggedebaqi, 2019-10-24:
@coolsnowwolf 多谢大佬的提示，编译成功了。

closed 2019-10-25 by coolsnowwolf