coolsnowwolf/lede #1843: SSR PLUS LUCI页面打不开了

tanghannb, 2019-10-22:
一点开就显示：
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute function dispatcher target for entry '/admin/services/shadowsocksr'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute cbi dispatcher target for entry '/admin/services/shadowsocksr/client'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/shadowsocksr/client.lua:18: bad argument #1 to 'upper' (string expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>
[20191022.txt](https://github.com/coolsnowwolf/lede/files/3756740/20191022.txt)

日志在附件，求各位看看，之前配置的还在，现在还能连接，就更了今天的源码

ledoen, 2019-10-22:
我今天编译的，第一次编译的固件刷上以后，能打开ssrplus界面，但是增加服务节点以后再点客户端界面就变成了跟你这个差不多的东西（没有保存），我怀疑是生成.config文件时，我只选了v2ray，并且勾掉了一些其他luci应用，第二次编译我把ssr所有功能勾上，并且其他东西没动，这一次ssrplus正常使用，没再出问题。

closed 2019-10-23 by coolsnowwolf

sanfeng0088, 2020-07-06:
和你的一样，刷好才半个小时
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute function dispatcher target for entry '/admin/services/shadowsocksr'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute cbi dispatcher target for entry '/admin/services/shadowsocksr/client'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/shadowsocksr/client.lua:18: bad argument #1 to 'upper' (string expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>