coolsnowwolf/lede #1844: 做旁路由FQ，为什么在用360的SO速度很慢呢？

hzdlive, 2019-10-23:
我安装好设置LAN后，在没有打开S-S R plus+ 的情况下
我就尝试用360浏览器的so来搜索资料，速度没有影响，
接下来，我打开S-S R plus +功能来FQ，我直接在地址栏里输入www.so.com，打开就有点慢，其他的比如www.baidu.com,www.bing.com速度没有影响
那么360浏览器的so速度就有点卡了。
我以为是S-S R plus+ 的影响，我就把S-S R plus + 关掉，没有想到
360浏览器的so功能打不开了，其他网页的链接能直打开。
我直接在地址栏里输入www.so.com，www.so.com打不开，其他的比如www.baidu.com,www.bing.com速度没有影响
我只好再把S-S R plus +功能打开，然后360浏览器的so又可以用了，只是速度有点卡，其他的网页打开速度正常。
我是用GFW列表模式。

是不是S-S R plus +的这个功能有BUG呢？

coolsnowwolf, 2019-10-23:
描述不清楚，不知道你想做什么，具体怎么配。多数是 DNS 配错了

closed 2019-10-23 by coolsnowwolf

hzdlive, 2019-10-23:
我是用爱快做主路由，lean做旁路由（运行ssr plus+）

hzdlive, 2019-10-23:
> 描述不清楚，不知道你想做什么，具体怎么配。多数是 DNS 配错了

我是用爱快做主路由，lean做旁路由（运行ssr plus+）
ssr plus+
运行模式:gfw列表模式
DNS解析方式:PDNSD
访问国外域名DNS服务器:8.8.4.4

接口里只有一个LAN
网关设爱快的内网地址
DNS:223.5.5.5,119.29.29.29