coolsnowwolf/lede #1856: SSR+模式设置与安卓手机app不能翻墙的问题

uoow, 2019-10-24:
环境：SSR+用的是V2RAY，路由：B70，MTK7621
问题：
1：GFW模式下所有流量都走梯子。
2：PC上浏览器可翻；安卓手机上用浏览器可翻，但是app不可翻。

请问要解决这两个问题要怎么解决？

coolsnowwolf, 2019-10-24:
用 iPhone 即可，或者别用国产手机

uoow, 2019-10-24:
iPhone是可以。国产安卓是无解吗？ @coolsnowwolf 
我也是MIUI10，但不行，谷歌商店、youtube的app都不可以。 @Her0R0cket

Her0R0cket, 2019-10-24:
你是不是原版系统?

uoow, 2019-10-24:
> 你是不是原版系统?

你这原版系统应该怎么理解？
就是小米的手机，小米的MIUI，安装的谷歌商店app

coolsnowwolf, 2019-10-24:
我的测试 APP 结果： iOS 都可以，小米8 刷最新 MIUI系统不可以，刷XDA 的 AOSP 改版系统 9.0 没任何问题

大家觉得是什么原因呢？

hcym, 2019-10-24:
我的杂牌安卓谷歌市场更新都用的路由模式，一直没问题

greennyreborn, 2019-10-24:
华为P30之前也有问题，netflix 和 youtube 都用不了，但是在手机上FQ后就可以正常使用了。
我猜测是手机系统劫持了dns解析

Her0R0cket, 2019-10-24:
我刚刚用mi4c新刷的稳定版10.1.1(7.0).在应用商店里安装Google play时,提示安装gms,成功安装后卡核对信息一次,第二次进去后成功登录,安装YouTube后提示更新gms,更新后使用正常
红米6有之前的数据,系统版本10.3.4.使用8.0版本的4件套,手动安装,安装youtube后更新gms,正常使用.

uoow, 2019-10-24:
> 我刚刚用mi4c新刷的稳定版10.1.1(7.0).在应用商店里安装Google play时,提示安装gms,成功安装后卡核对信息一次,第二次进去后成功登录,安装YouTube后提示更新gms,更新后使用正常
> 红米6有之前的数据,系统版本10.3.4.使用8.0版本的4件套,手动安装,安装youtube后更新gms,正常使用.
> [image]

那看来可以closed了

closed 2019-10-24 by uoow