coolsnowwolf/lede #1864: ssr+新版出现“daemon.err pdnsd[601]: Could not listen on tcp socket: Not a socket”！！！

zxlhhyccc, 2019-10-24:
Thu Oct 24 22:22:51 2019 daemon.info pdnsd[601]: pdnsd-1.2.9b-par starting.
Thu Oct 24 22:22:51 2019 daemon.err pdnsd[601]: error in UDP recv: Not a socket
Thu Oct 24 22:22:51 2019 daemon.err pdnsd[601]: Could not listen on tcp socket: Not a socket
Thu Oct 24 22:22:51 2019 daemon.err pdnsd[601]: error in UDP recv: Not a socket
Thu Oct 24 22:22:51 2019 daemon.err pdnsd[601]: error in UDP recv: Not a socket
Thu Oct 24 22:22:51 2019 daemon.err pdnsd[601]: error in UDP recv: Not a socket
Thu Oct 24 22:22:51 2019 daemon.err pdnsd[601]: error in UDP recv: Not a socket
Thu Oct 24 22:22:52 2019 daemon.err pdnsd[601]: error in UDP recv: Not a socket
Thu Oct 24 22:22:52 2019 daemon.err pdnsd[601]: error in UDP recv: Not a socket
Thu Oct 24 22:22:52 2019 daemon.err pdnsd[601]: error in UDP recv: Not a socket
Thu Oct 24 22:22:52 2019 daemon.err pdnsd[601]: error in UDP recv: Not a socket
Thu Oct 24 22:22:52 2019 daemon.err pdnsd[601]: error in UDP recv: Not a socket
Thu Oct 24 22:22:52 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Oct 24 22:22:52 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Oct 24 22:22:53 2019 kern.info /usr/bin/ssr-redir[550]: tcp port reuse enabled
Thu Oct 24 22:22:53 2019 kern.info /usr/bin/ssr-redir[550]: listening at 0.0.0.0:1234
Thu Oct 24 22:22:53 2019 kern.info /usr/bin/ssr-redir[550]: running from root user
Thu Oct 24 22:22:54 2019 kern.info /usr/bin/ssr-redir[548]: tcp port reuse enabled
Thu Oct 24 22:22:54 2019 kern.info /usr/bin/ssr-redir[548]: listening at 0.0.0.0:1234
Thu Oct 24 22:22:54 2019 kern.info /usr/bin/ssr-redir[548]: running from root user
Thu Oct 24 22:22:54 2019 kern.info /usr/bin/ssr-redir[583]: UDP relay enabled
Thu Oct 24 22:22:54 2019 kern.info /usr/bin/ssr-redir[583]: udp port reuse enabled
Thu Oct 24 22:22:54 2019 kern.info /usr/bin/ssr-redir[583]: TCP relay disabled
Thu Oct 24 22:22:54 2019 kern.info /usr/bin/ssr-redir[583]: listening at 0.0.0.0:1234
Thu Oct 24 22:22:54 2019 kern.info /usr/bin/ssr-redir[583]: running from root user

outemancut, 2019-10-25:
pdnsd的问题！

closed 2019-10-25 by coolsnowwolf

zxlhhyccc, 2019-10-25:
> pdnsd的问题！

有办法解决吗？还是要用其他的替代？