coolsnowwolf/lede #1867: 6174网卡启动报错。。。

adaaaaaa, 2019-10-25:
`[    6.663942] NET: Registered protocol family 24
[    6.665311] usbcore: registered new interface driver snd-usb-audio
[    6.698080] ath10k_pci 0000:00:10.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[    7.094308] ath10k_pci 0000:00:10.0: Direct firmware load for ath10k/pre-cal-pci-0000:00:10.0.bin failed with error -2
[    7.096408] ath10k_pci 0000:00:10.0: Falling back to user helper
[   69.802432] ath10k_pci 0000:00:10.0: Direct firmware load for ath10k/cal-pci-0000:00:10.0.bin failed with error -2
[   69.804542] ath10k_pci 0000:00:10.0: Falling back to user helper
[   85.801079] random: crng init done
[  131.280291] ath10k_pci 0000:00:10.0: qca6174 hw3.2 target 0x05030000 chip_id 0x00340aff sub 1a56:1535
[  131.282403] ath10k_pci 0000:00:10.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 1
[  131.285916] ath10k_pci 0000:00:10.0: firmware ver WLAN.RM.4.4.1-00140-QCARMSWPZ-1 api 6 features wowlan,ignore-otp,mfp crc32 29eb8ca1
[  131.358881] ath10k_pci 0000:00:10.0: board_file api 2 bmi_id N/A crc32 4ed3569e
[  131.927500] ath10k_pci 0000:00:10.0: Unknown eventid: 118809
[  131.930509] ath10k_pci 0000:00:10.0: Unknown eventid: 90118
[  131.932582] ath10k_pci 0000:00:10.0: htt-ver 3.60 wmi-op 4 htt-op 3 cal otp max-sta 32 raw 0 hwcrypto 1
[  132.052798] ath: EEPROM regdomain: 0x69
[  132.052798] ath: EEPROM indicates we should expect a direct regpair map
[  132.052799] ath: Country alpha2 being used: 00
[  132.052799] ath: Regpair used: 0x69
`

哪位老哥看出什么问题了不？？
要2分多钟才能启动op完毕。。。以前都是几秒的。。。

adaaaaaa, 2019-11-23:
现在的版本也是这样，但是，我卸载了autosamba和samba3相关的所有插件之后，就没事了。。。是因为有冲突吗？？🙃🙃🙃

closed 2019-12-24 by coolsnowwolf

malash, 2022-05-27:
@coolsnowwolf 我在现在最新的master分支也遇到了一样的问题，确实是autosamba/samba3与ath10k不太兼容，如果开启samba会导致ath10k_pci花费将近2分钟时间进行firmware的初始化。我的网卡是Killer 1535（QCA6174）。

我建议re-open这个issue，或者向上游反馈。

开启samba：

```
[    7.808036] ath10k_pci 606c:00:00.0: can't derive routing for PCI INT A
[    7.808687] ath10k_pci 606c:00:00.0: PCI INT A: no GSI
[    7.810928] ath10k_pci 606c:00:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[  127.474815] ath10k_pci 606c:00:00.0: qca6174 hw3.2 target 0x05030000 chip_id 0x00340aff sub 1a56:1535
[  127.475529] ath10k_pci 606c:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[  127.478067] ath10k_pci 606c:00:00.0: firmware ver WLAN.RM.4.4.1-00157-QCARMSWPZ-1 api 6 features wowlan,ignore-otp,mfp crc32 90eebefb
[  127.557799] ath10k_pci 606c:00:00.0: board_file api 2 bmi_id N/A crc32 706c395e
[  127.653186] ath10k_pci 606c:00:00.0: htt-ver 3.60 wmi-op 4 htt-op 3 cal otp max-sta 32 raw 0 hwcrypto 1
[  127.735987] ath: EEPROM regdomain: 0x6c
[  127.735993] ath: EEPROM indicates we should expect a direct regpair map
[  127.735994] ath: Country alpha2 being used: 00
[  127.735995] ath: Regpair used: 0x6c
[  127.736372] ath10k_pci 606c:00:00.0 wlan0: no netdev found for vf serial:25023
[  127.751061] usbcore: registered new interface driver ath9k_htc
[  129.422315] ath: EEPROM regdomain: 0x8348
[  129.422324] ath: EEPROM indicates we should expect a country code
[  129.422325] ath: doing EEPROM country->regdmn map search
[  129.422327] ath: country maps to regdmn code: 0x3a
[  129.422328] ath: Country alpha2 being used: US
[  129.422329] ath: Regpair used: 0x3a
[  129.422330] ath: regdomain 0x8348 dynamically updated by user
[  130.464129] ath10k_pci 606c:00:00.0 wlan0: no netdev found for vf serial:25023
```

关闭samba：

```
[    7.811395] ath10k_pci 606c:00:00.0: can't derive routing for PCI INT A
[    7.812062] ath10k_pci 606c:00:00.0: PCI INT A: no GSI
[    7.814253] ath10k_pci 606c:00:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[    8.136634] ath10k_pci 606c:00:00.0: qca6174 hw3.2 target 0x05030000 chip_id 0x00340aff sub 1a56:1535
[    8.137337] ath10k_pci 606c:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[    8.139533] ath10k_pci 606c:00:00.0: firmware ver WLAN.RM.4.4.1-00288- api 6 features wowlan,ignore-otp,mfp crc32 bf907c7c
[    8.219221] ath10k_pci 606c:00:00.0: board_file api 2 bmi_id N/A crc32 62812cba
[    8.312429] ath10k_pci 606c:00:00.0: htt-ver 3.87 wmi-op 4 htt-op 3 cal otp max-sta 32 raw 0 hwcrypto 1
[    8.390203] ath: EEPROM regdomain: 0x6c
[    8.390209] ath: EEPROM indicates we should expect a direct regpair map
[    8.390211] ath: Country alpha2 being used: 00
[    8.390212] ath: Regpair used: 0x6c
[    8.390599] ath10k_pci 606c:00:00.0 wlan0: no netdev found for vf serial:25023
[    8.403664] usbcore: registered new interface driver ath9k_htc
[   25.327188] ath: EEPROM regdomain: 0x8348
[   25.327193] ath: EEPROM indicates we should expect a country code
[   25.327194] ath: doing EEPROM country->regdmn map search
[   25.327195] ath: country maps to regdmn code: 0x3a
[   25.327196] ath: Country alpha2 being used: US
[   25.327197] ath: Regpair used: 0x3a
[   25.327197] ath: regdomain 0x8348 dynamically updated by user
[   26.367106] ath10k_pci 606c:00:00.0 wlan0: no netdev found for vf serial:25023
```